stepjam/PyRep #2: Getting 'The 'MeshCalc' plugin could not be initialized.'

denoza, 2019-06-27:
I tried to run `example_reinforcement_learning_env.py` but I am getting this error: `The 'MeshCalc' plugin could not be initialized`. I am not running the example in headless mode. While launching Vrep manually, everything works fine and the `MeshCalc` plugin loads fine. Is there something wrong with my Config? I am using Vrep 3.6.2 and I followed the README for installation of PyRep.

stepjam, 2019-06-27:
Hi denoza,
What version of Linux are you using?

denoza, 2019-06-27:
I am using Ubuntu 16.04

stepjam, 2019-06-27:
Thanks. I was able to reproduce on 16.04. Will try and resolve asap.

stepjam, 2019-06-27:
Update: 
Still trying to figure out whats going wrong on Ubuntu 16.04. When using a 16.04 in a virtual machine, it seems to run fine, but when running on bare metal, we get the error. Very strange.

Coppelia, 2019-06-27:
What error message does the console give, when V-REP tries to load the meshCalc plugin? e.g.:
Simulator launched.
Plugin 'MeshCalc': loading...
Plugin 'MeshCalc': load succeeded.

stepjam, 2019-06-27:
On my bare-metal installation I have -3 returned from 'simLoadModule_internal'. Which gives the error:
'The plugin probably couldn't load dependency libraries. For additional infos, modify the script 'libLoadErrorCheck.sh', run it and inspect the output.'

But when running 'ldd', it all looks fine, which is not too surprising given that V-REP loads fine if you are launching from the c++ client. What's also strange is that all the other plugins load fine.

Coppelia, 2019-06-27:
That is very strange...
Can you try with the lib that I have recompiled here?: http://www.coppeliarobotics.com/files/tmp/libv_repExtMeshCalc.so

stepjam, 2019-06-27:
That did it!
What did you modify?

Coppelia, 2019-06-27:
I simply recompiled the plugin, which was still the version from 2016!!!

stepjam, 2019-06-27:
OK great! Would it be possible to repackage V-REP with recompiled version?

In the meantime, @denoza (and other Ubuntu 16 users), please put the following file in the V-REP install directory: http://www.coppeliarobotics.com/files/tmp/libv_repExtMeshCalc.so

Once the recompiled version has been repackaged, I will close this issue.

Coppelia, 2019-06-27:
done!

stepjam, 2019-06-27:
Great! Thanks Marc!

closed 2019-06-27 by stepjam

denoza, 2019-06-27:
Thanks for the prompt resolution of this issue @stepjam and @Coppelia