stepjam/PyRep #6: Improvement Idea: Trigger multiple simulation steps at once

Phlogiston90, 2019-07-01:
Hi Stephen, 

in my particular RL problem, I want to simulate 20 simulation steps before the agent observes the new state, because it drops something and has to see where it lands, which takes longer than 1 simulation step. Increasing the simulation time delta by 20 times obviously doesn't work as this will not consider collisions that might happen in between. 
I could imagine that I would not be the only one that has a setting like this. 

Sending 20 times PyRep.step() is slow (~210ms in my scene), which is the same time as sending 20 times  vrep.simxSynchronousTrigger() in the original VREP python API.
I was able to speed this up by a factor of ~10 times by writing a customization script that executes 20 simulation steps at one call as suggested in http://www.forum.coppeliarobotics.com/viewtopic.php?f=9&t=7926 .
One would have to write a customization script containing
```
function sysCall_init()
    stepsToRun=0
end

function start20StepRun()
    stepsToRun=20
end

function sysCall_beforeMainScript()
    return {doNotRunMainScript=(stepsToRun==0)}
end

function sysCall_actuation()
    stepsToRun=stepsToRun-1
end
```
and then don't use the synchronous mode in VREP.
Unfortunately, this doesn't work with PyRep anymore, and I was wondering if you could add this functionality directly into it, e.g. that PyRep.step() accepts the number of steps as a parameter. This would be an awesome functionality and another bonus for PyRep!

Best, 
Mark

stepjam, 2019-07-01:
Sounds like a good addition to me. Expect to see this in the next few hours.

stepjam, 2019-07-02:
I've looked in to this more and I think I have a plan. I'll have to make the modifications to the V-REP source, so it will take some time before this feature becomes available (in the next public revision of V-REP).

Phlogiston90, 2019-07-02:
That's awesome!!

closed 2020-07-04 by stale[bot]