stepjam/PyRep #10: Textures not loaded on second simulation start in headless mode

Phlogiston90, 2019-07-02:
While the first image is with textures, the second image is without textures when executing the following:

```
pr = PyRep()
pr.launch(SCENE_PATH, headless=True)
pr.start()
pr.step()
cam = VisionSensor("camera")
img = cam.capture_rgb()
plt.imshow(img)
plt.show()
pr.stop()

pr.start()
pr.step()
img = cam.capture_rgb()
plt.imshow(img)
plt.show()
pr.stop()
pr.shutdown()
```

stepjam, 2019-07-02:
I can't seem to reproduce this. Could you perhaps include a link to the scene file you are using?

Phlogiston90, 2019-07-02:
[texture_test.zip](https://github.com/stepjam/PyRep/files/3349192/texture_test.zip)

stepjam, 2019-07-02:
Thanks. Yes I am now getting the same issue. Will look in to this.

stepjam, 2019-07-02:
If you swap to the 'External Renderer' or 'OpenGL 3' renderer then this issue does not arise. This may be a problem with V-REP rather than PyRep. @Coppelia any ideas?

Phlogiston90, 2019-07-02:
Thanks a lot! That works indeed.

Coppelia, 2019-07-09:
Thanks for the info. You can probably make it work by setting variable _oglCompatibilityTweak1_ to true in _system/usrset.txt_.
This will be changed for next revision.
Cheers

closed 2019-07-09 by Coppelia