stepjam/PyRep #11: How to open 2 scenes

Phlogiston90, 2019-07-02:
Hi Stephen,

how do I open 2 scenes (or the same scene twice) to have different environments for RL?

The following does not work, neither in headless mode nor in GUI mode, and neither with having the same PyRep instance (pr) or two (pr and pr2) for launching the scene files.
Also I noted that after a .launch(), `os.path.join(os.path.dirname(os.path.abspath(__file__))` is in the V-REP installation folder (so I copied all scenes there as well).

```
from pyrep import PyRep
from pyrep.objects.vision_sensor import VisionSensor
import os
from matplotlib import pyplot as plt

if __name__ == "__main__":
    
    SCENE_PATH1 = os.path.join(os.path.dirname(os.path.abspath(__file__)), "texture_test.ttt")
    SCENE_PATH2 = os.path.join(os.path.dirname(os.path.abspath(__file__)), "texture_test_copy.ttt")
    print(SCENE_PATH1, SCENE_PATH2)
    
    pr = PyRep()
    pr.launch(SCENE_PATH1, headless=True)
    cam = VisionSensor("Cam")
    pr.start()
    pr.step()
    img = cam.capture_rgb()
    plt.imshow(img)
    plt.show()
    pr.stop()
  
    pr2 = PyRep()
    pr2.launch(SCENE_PATH2, headless=True)
    cam = VisionSensor("Cam")
    pr2.start()
    pr2.step()
    img = cam.capture_rgb()
    plt.imshow(img)
    plt.show()
    pr2.stop()
```
Best, 
Mark

stepjam, 2019-07-02:
Each PyRep instance needs to be on a different process. You will need to use the Python [multiprocessing](https://docs.python.org/3.6/library/multiprocessing.html) module. I can put an example of this in the README.

closed 2019-07-02 by Phlogiston90

stepjam, 2019-07-02:
Multiprocessing example now added to README.