stepjam/PyRep #12: Improvement Idea/Bug: Pass SIGINT to PyRep thread

Phlogiston90, 2019-07-02:
Hi Stephen, 

I can only quit PyRep by sending a SIGKILL, it would be nice if would react to SIGINT (CTRL+C) or at least SIGTERM (end signal of debugger)...

Best,
Mark

stepjam, 2019-07-02:
You can use SIGTERM (ctrl-\) instead. I can't seem to get it to react to the SIGINT signals.

closed 2019-07-02 by stepjam

Phlogiston90, 2019-07-02:
SIGTERM doesn't always work, but I cannot tell exactly when yet. I will look into this and try to find out why, it could also be because of other libraries like PyTorch etc.

hnyu, 2022-03-02:
Hi @Phlogiston90 , I wonder whether this issue has been fixed? Right now I also have the problem of not being able to kill the process by sending SIGINT.

tonyddg, 2025-02-12:
It seems that the default handler was changed (In default handler, a [KeyboardInterrupt](https://docs.python.org/3.13/library/signal.html#signal.SIGINT) will be raised). You can handle SIGINT by the module [signal](https://docs.python.org/3.13/library/signal.html#module-signal) to stop simulate looping instead of passing it to thread like the flowing minimum example:

```python
from pyrep import PyRep
import signal

pr = PyRep()
pr.launch('scene/to/launch') 
pr.start()  # Start the simulation

is_running = True
# Register a handler to change the running flag
def sig_handle(sig, frame):
    global is_running
    is_running = False
    print(f"Force Exit For Signal {sig}")
signal.signal(signal.SIGINT, sig_handle)
signal.signal(signal.SIGTERM, sig_handle)

# The simulate loop is kept by the running flag
while is_running:
    pr.step()

pr.stop()  # Stop the simulation
pr.shutdown()  # Close the application
```