stepjam/PyRep #8: capture_rgb() returns an empty image

denoza, 2019-07-02:
I am trying to run a simulation in headless mode in which my robot captures an image from the scene and save it to the disk via opencv. My code is as follows:

`pr.launch(SCENE_FILE, headless=True)`
`pr.start()`
`agent = UR3()`
`agent.set_joint_positions(initialPos)`
`pr.step()`
`cam = VisionSensor('cam')`
`img = cam.capture_rgb()`
`img = np.uint8(img)`
`target_image = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)`
`cv2.imwrite('targ.png', self.target_image)`

However, as a result, the image saved is an empty image with all zeros. How can I get scene image from the simulation?

P.S. I tried it with `headless=False` and the result was same.

Phlogiston90, 2019-07-02:
You have to write
```
img = np.uint8(img * 256.)
```
then it works.

stepjam, 2019-07-02:
Yes that is correct. The function returns a float matrix in the range 0 - 1; so you will need to scale before casting. I will update the doc-string to make this clear :).

closed 2019-07-02 by stepjam