stepjam/PyRep #21: error: command 'x86_64-linux-gnu-gcc' failed not solved with python3-dev updated

quantumiracle, 2019-07-13:
Hi Stephen,

I tried to install the PyRep, and meet the error as follows:

`x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.6/build/pyrep/backend/pyrep/backend/_v_rep_cffi.o -L/home/quantumiracle/research/vrep/PyRep/PyRep -lv_rep -o build/lib.linux-x86_64-3.6/pyrep/backend/_v_rep_cffi.cpython-36m-x86_64-linux-gnu.so
/usr/bin/ld: cannot find -lv_rep
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1`

And I tried `sudo apt-get install python3-dev` and it's already the newest, so it doesn't solved my case. 

Moreover, if I tried to install it in a conda environment with python3, the error would be:
`/usr/bin/x86_64-linux-gnu-ld: cannot find -lv_rep
collect2: error: ld returned 1 exit status
error: command 'gcc' failed with exit status 1
`

Do you know how to solve this?

Thanks,
Hanry

stepjam, 2019-07-13:
Hi Hanry,

It seems it is looking for libv_rep.so in `-L/home/quantumiracle/research/vrep/PyRep/PyRep`.
Can you confirm that is where V-REP is installed? Your path name does not seem to suggest this.

quantumiracle, 2019-07-13:
Thanks, Stephen.

My fault, I used the PyRep's path instead of the Vrep path. It works now!

closed 2019-07-13 by quantumiracle