stepjam/PyRep #29: do you have a plan to support windows?

weihuan, 2019-07-29:


stepjam, 2019-07-29:
Hi,
I haven't actually tried running on windows (we don't have one in the lab)... So you may find that it works; but no, I did not plan to support windows or mac. Let me know what issues you get when trying to install.

stepjam, 2019-08-01:
@weihuan Any updates on this? If not, I will close the issue for now.

stepjam, 2019-08-03:
Closed due to inactivity

closed 2019-08-03 by stepjam

bknight-i3drobotics, 2020-03-24:
I am also trying to build this for windows as it would be incredibly useful. I have not used cffi before and I'm running into an issue where it's looking for coppeliaSim.lib but the library is shared not static so it should be looking for coppeliaSim.dll. Just wondered if you had any advice on how to compile with cffi using a shared library.