stepjam/PyRep #30: suppress_std_out_and_err isn't thread safe

dougsm, 2019-07-30:
Hi Stephen,
I've had a few issues with `suppress_std_out_and_err` ([link](https://github.com/stepjam/PyRep/blob/0c964caebc4c3a0bfae31725fddadd0405a68dc7/pyrep/backend/utils.py#L64)) and threading.  When trying to parallelise, for example, simulation and  training, it's possible for a `print` or logger in the other thread to write to the closed stdout, which causes python to crash.  
I had a look but haven't been able to come up with a good way to avoid the possible race condition.  Any ideas?

stepjam, 2019-07-30:
Hi Doug,
I knew that hack would come back to haunt me ...
A very quick fix is to simply remove the `suppress_std_out_and_err`. That unfortunately means that you will get some OMPL logs spewed to the console (even though we request no logging). I may have to get in contact with Mark and see if we can find a longer term solution to this...

dougsm, 2019-07-30:
Haha, thanks. Yeah, my current work around is to just remove it and put up with the output, but obviously thats not ideal.

stepjam, 2019-07-31:
This is not a priority for now, so I'll close with the intention to look at a longer term solution further down the line.

closed 2019-07-31 by stepjam

robfiras, 2020-11-11:
It seems like were are not getting any OMPL logs anymore when removing `suppress_std_out_and_err`. Maybe we can remove it? Because I am having some trouble with it, when doing multiprocessing.