stepjam/PyRep #31: OpenAI Gym

AI-Guru, 2019-07-30:
Namaste!

Wow! This looks very promising!

Do you intend to make PyRep OpenAI-Gym compatible? That would open doors to a lot of Deep Reinforcement Learning.

https://gym.openai.com

stepjam, 2019-07-30:
Hi,
Yes there is an on-going project that will be complementary to OpenAI Gym. Expected release is about 2 months.

In the meantime, it's super simple to use PyRep as a reinforcement learning environment. Look at the [example here](https://github.com/stepjam/PyRep/blob/master/examples/example_reinforcement_learning_env.py) :)

closed 2019-07-30 by stepjam

stepjam, 2019-10-05:
The project I mentioned is now live: [https://github.com/stepjam/RLBench](https://github.com/stepjam/RLBench).