stepjam/PyRep #32: Cannot mix incompatible Qt library (version 0x50907) with this library (version 0x50c01) Aborted (core dumped)

sandipan1, 2019-07-31:
When I run python example_youbot_navigation.py , I get the error 
```V-REP PRO EDU V3.6.2. (rev. 0)
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
GLib-GIO-Message: 20:32:08.861: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
Using the default Lua library.
Loaded the video compression library.
Add-on script 'vrepAddOnScript-addOnScriptDemo.lua' was loaded.
Add-on script 'vrepAddOnScript-b0RemoteApiServer.lua' was loaded.
Add-on script 'vrepAddOnScript_PyRep.lua' was loaded.
If V-REP crashes now, try to install libgl1-mesa-dev on your system:
>sudo apt install libgl1-mesa-dev
OpenGL: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2) , Version: 3.0 Mesa 18.0.5
...did not crash.
Simulator launched.
Plugin 'MeshCalc': loading...
Plugin 'MeshCalc': load succeeded.
Plugin 'Assimp': loading...
Plugin 'Assimp': warning: replaced variable 'simAssimp'
Plugin 'Assimp': load succeeded.
Plugin 'BlueZero': loading...
Plugin 'BlueZero': warning: replaced variable 'simB0'
Plugin 'BlueZero': load succeeded.
Plugin 'BubbleRob': loading...
Plugin 'BubbleRob': load succeeded.
Plugin 'Bwf': loading...
Plugin 'Bwf': load succeeded.
Plugin 'CodeEditor': loading...
Plugin 'CodeEditor': load succeeded.
Plugin 'Collada': loading...
Plugin 'Collada': load succeeded.
Plugin 'ConvexDecompose': loading...
Plugin 'ConvexDecompose': load succeeded.
Plugin 'CustomUI': loading...
Plugin 'CustomUI': warning: replaced variable 'simUI'
Plugin 'CustomUI': warning: replaced function 'simUI.insertTableRow@CustomUI'
Plugin 'CustomUI': warning: replaced function 'simUI.removeTableRow@CustomUI'
Plugin 'CustomUI': warning: replaced function 'simUI.insertTableColumn@CustomUI'
Plugin 'CustomUI': warning: replaced function 'simUI.removeTableColumn@CustomUI'
Plugin 'CustomUI': warning: replaced function 'simUI.setScene3DNodeParam@CustomUI'
Plugin 'CustomUI': load succeeded.
Plugin 'DynamicsBullet-2-78': loading...
Plugin 'DynamicsBullet-2-78': load succeeded.
Plugin 'DynamicsBullet-2-83': loading...
Plugin 'DynamicsBullet-2-83': load succeeded.
Plugin 'DynamicsNewton': loading...
Plugin 'DynamicsNewton': load succeeded.
Plugin 'DynamicsOde': loading...
Plugin 'DynamicsOde': load succeeded.
Plugin 'DynamicsVortex': loading...
Plugin 'DynamicsVortex': load succeeded.
Plugin 'ExternalRenderer': loading...
Plugin 'ExternalRenderer': load succeeded.
Plugin 'ICP': loading...
Plugin 'ICP': warning: replaced variable 'simICP'
Plugin 'ICP': load succeeded.
Plugin 'Image': loading...
Plugin 'Image': warning: replaced variable 'simIM'
Plugin 'Image': load succeeded.
Plugin 'K3': loading...
Plugin 'K3': load succeeded.
Plugin 'LuaCommander': loading...
Plugin 'LuaCommander': warning: replaced variable 'simLuaCmd'
Plugin 'LuaCommander': load succeeded.
Plugin 'LuaRemoteApiClient': loading...
Plugin 'LuaRemoteApiClient': load succeeded.
Plugin 'Mtb': loading...
Plugin 'Mtb': load succeeded.
Plugin 'OMPL': loading...
Plugin 'OMPL': warning: replaced variable 'simOMPL'
Plugin 'OMPL': load succeeded.
Plugin 'OpenGL3Renderer': loading...
Plugin 'OpenGL3Renderer': load succeeded.
Plugin 'OpenMesh': loading...
Plugin 'OpenMesh': load succeeded.
Plugin 'Qhull': loading...
Plugin 'Qhull': load succeeded.
Plugin 'RRS1': loading...
Plugin 'RRS1': load succeeded.
Plugin 'ReflexxesTypeII': loading...
Plugin 'ReflexxesTypeII': load succeeded.
Plugin 'RemoteApi': loading...
Starting a remote API server on port 19997
Plugin 'RemoteApi': load succeeded.
Plugin 'RosInterface': loading...
Error with plugin 'RosInterface': load failed (could not load). The plugin probably couldn't load dependency libraries. For additional infos, modify the script 'libLoadErrorCheck.sh', run it and inspect the output.
Plugin 'SDF': loading...
Plugin 'SDF': warning: replaced variable 'simSDF'
Plugin 'SDF': load succeeded.
Plugin 'SurfaceReconstruction': loading...
Plugin 'SurfaceReconstruction': warning: replaced variable 'simSurfRec'
Plugin 'SurfaceReconstruction': load succeeded.
Plugin 'Urdf': loading...
Plugin 'Urdf': load succeeded.
Plugin 'Vision': loading...
Plugin 'Vision': load succeeded.
Using the 'MeshCalc' plugin.
Initializing the Bullet physics engine in plugin 'DynamicsBullet_2_83'...
Engine version: 2.83.7
Plugin version: 10
Initialization successful.
Cannot mix incompatible Qt library (version 0x50907) with this library (version 0x50c01)
Aborted (core dumped)
```

qmake --version  outputs 
```(python_robotics) sandipan $~ qmake --version
QMake version 3.1
Using Qt version 5.9.7 in /home/sandipan/miniconda3/envs/python_robotics/lib
```
qtchooser -l outputs
```
(python_robotics) sandipan $~ qtchooser -l
4
5
default
qt4-x86_64-linux-gnu
qt4
qt5-x86_64-linux-gnu
qt5
```

stepjam, 2019-07-31:
Hi,

Can you confirm that you are able to run V-REP as normal from the command line? (i.e. by running `./vrep.sh` from the V-REP root).

sandipan1, 2019-07-31:
Yes vrep is running fine independently

stepjam, 2019-07-31:
Thanks. And I'm assuming then that other examples will also crash with a similar message?

sandipan1, 2019-07-31:
No I am facing this issue particularly on example_youbot_navigation.py

stepjam, 2019-08-01:
Interesting...
OK, let's start diagnosing. Could you run your debugger and let me know what line it dies at? 
Thanks.

sandipan1, 2019-08-01:
The error is at pr.step()
```
> /home/sandipan/robotics/PyRep/examples/example_youbot_navigation.py(45)<module>()
     44         done = path.step()
---> 45         pr.step()
     46 

ipdb>                                                                                     
Cannot mix incompatible Qt library (version 0x50905) with this library (version 0x50c01)
Aborted (core dumped)
```

Specifically at this line in vrep.py
```
> /home/sandipan/.local/lib/python3.7/site-packages/pyrep/backend/vrep.py(53)simExtStep()
     52 def simExtStep(stepIfRunning=True):
---> 53     lib.simExtStep(stepIfRunning)
     54 

ipdb>                                                                                     
Cannot mix incompatible Qt library (version 0x50905) with this library (version 0x50c01)
Aborted (core dumped)
```

sandipan1, 2019-08-01:
Hi 
The problem is solved when I installed pyrep inside a different conda env or make a fresh install with the default python3 . 
Thanks for the support and sorry for the trouble.

closed 2019-08-01 by sandipan1