stepjam/PyRep #33: Set control frequency to 1khz

elieJalbout, 2019-08-05:
Is there anyway to change the control frequency to 1khz within pyrep? and how

stepjam, 2019-08-07:
Hi,
Could you clarify what you mean? 
Are you saying you want to step the physics simulation at 1kHz? Or do you want to decrease the simulation step size? If it is the latter, then you can use the function: `set_simulation_timestep(1.0)`. [Doc here](https://pyrep.readthedocs.io/en/latest/pyrep.html?highlight=set_simulation_timestep#pyrep.pyrep.PyRep.set_simulation_timestep).

elieJalbout, 2019-08-08:
Hi, 
Thanks for the quick support. Well, I meant the latter, decreasing simulation step size. I already tried set_simulation_timestep(1.0) but it still says in V-Rep 's GUI that simulation step size dt=50ms. Here's my code:

self.pr = PyRep()

self.pr.launch(get_scene_file('scene_reinforcement_learning_env.ttt'), headless=False)
self.pr.start()\
self.pr.set_simulation_timestep(1.0)

stepjam, 2019-08-08:
Hi. To see the changes in the UI do `self.pr.step_ui()` after setting the timestep.

dougsm, 2019-08-15:
@elieJalbout A few things to note:

1. Make sure you call set_simulation_timestep *before* you call pr.start() or it won't take effect.
2. The VREP GUI shows the dt in two places. As long as the on screen display (i.e. where mine says dt=50.0ms) is correct after calling set_simulation_timestep then it's working.  For some reason the dropdown bar in the menu (i.e. where it says dt=200ms) doesn't update.
[image]
3. `set_simulation_timestep` takes the time step in seconds, not milliseconds, so 1kHz would be 0.001. (@stepjam maybe clarify the documentation? This stumped me for a bit.).

stepjam, 2019-08-15:
Thanks @dougsm.
Yes you are right, it is indeed in seconds and not milliseconds. Will update the doc!

commit 5d93e8e5 (2019-08-15): Resolves #33. Improved docs.
closed 2019-08-15 by stepjam