stepjam/PyRep #34: Support Qt 5 GUI in Python

vincent-hui, 2019-08-07:
It would be great if PyRep can support [Qt 5 GUI in Python](https://www.qt.io/qt-for-python). 
Would you mind creating a simple example using PyRep and PySide2 ?

stepjam, 2019-08-07:
What do you mean by supporting QT 5 GUI?
What exactly do you have in mind?

vincent-hui, 2019-08-07:
http://www.coppeliarobotics.com/helpFiles/en/customUIPlugin.htm
It would be great if QtQuick is supported too.

stepjam, 2019-08-07:
I see. I consider this out of scope for what I had in mind for PyRep. PyRep is meant as a bridge between V-REP and your machine learning algorithm/dataset generator. If you want to add in new UI dialogs to VREP, then you should be instead using the plugin that you pointed to.

closed 2019-08-07 by stepjam

vincent-hui, 2019-08-07:
Qt can be used not only for VREP custom GUI, but also for machine learning algorithms.
One of Qt applications with MXNet https://youtu.be/UwJxLztoI1o
Imagine those robotic arms are being simulated in VREP instead of real robotic arms