stepjam/PyRep #36: ImportError when running in anaconda env

pointW, 2019-08-20:
When I try to use PyRep in an anaconda env, I got this error:

```Traceback (most recent call last):
  File "example_baxter_pick_and_pass.py", line 9, in <module>
    from pyrep import PyRep
  File "/home/dian/.local/lib/python3.7/site-packages/pyrep/__init__.py", line 1, in <module>
    from .pyrep import PyRep
  File "/home/dian/.local/lib/python3.7/site-packages/pyrep/pyrep.py", line 1, in <module>
    from pyrep.backend import vrep, utils
  File "/home/dian/.local/lib/python3.7/site-packages/pyrep/backend/vrep.py", line 2, in <module>
    from ._v_rep_cffi import ffi, lib
ImportError: /home/dian/Software/V-REP_PRO_EDU_V3_6_2_Ubuntu16_04/libv_rep.so.1: symbol _ZN17QAbstractItemView11eventFilterEP7QObjectP6QEvent, version Qt_5 not defined in file libQt5Widgets.so.5 with link time reference
```

It works fine if I install in system python interpreter. Does PyRep support working in anaconda now?

stepjam, 2019-08-20:
Hi there,
Could you first take a look at issue #32 and see if that solves your problem? If not, you can report back here and I'll look in to this further :).

pointW, 2019-08-21:
Hi,
Thanks for replying! I tried to install in a new conda env, but it still doesn't work. Though it works in system python interpreter, I would prefer to use conda since it will be easier for me to integrate pyrep with my current project.

I think the problem might be that in conda env, it is using a different Qt version created by anaconda.
When I do `qmake -version` inside a conda env, I got:
```
QMake version 3.1
Using Qt version 5.9.7 in /home/dian/anaconda3/lib
```
But If I do it in outside the conda env, I got:
```
QMake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu
```

stepjam, 2019-08-23:
I'm not an expert on conda I'm afraid.
It could be that `QT_QPA_PLATFORM_PLUGIN_PATH` needs to be redefined when using conda? Or perhaps the conda qt could be updated as it is missing some packages?

stepjam, 2019-08-28:
Hi,
Any updates with this?

stepjam, 2019-09-05:
Closing due to inactivity.

closed 2019-09-05 by stepjam