stepjam/PyRep #37: DogBot - quadruped robot simulation

g2re, 2019-08-21:
Link to URDF for DogBot: https://github.com/ReactRobotics/DogBotV4

stepjam, 2019-08-23:
Hi,
Thanks for the feature request; I cant give an exact ETA for this. I will probably be able to add this in once ICRA deadline has passed. Will leave this issue open until there are some update :)

g2re, 2019-08-29:
Cool!! Not long to go - good luck with the paper.

stepjam, 2019-12-14:
Hi @g2re. Unfortunately, I never got time to get around to this. Now with my time to split between maintaining [RLBench](https://github.com/stepjam/RLBench) and PyRep, I wont get a chance to implement new robots myself. Though I do encourage the community to continue to add. There is a tutorial on how to do so [here](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md). Perhaps you could try, or know someone who could bring in dogbot :smiley: .

closed 2019-12-14 by stepjam