stepjam/PyRep #38: How to contribute in adding new Robots ?

80avin, 2019-08-27:
How can we contribute in adding new robots ?
I saw, they're just ttm files which you've mentioned to be modified.
You can allow the opensource community to contribute in that, by sharing the required details.

stepjam, 2019-08-28:
Hi 80avin,
Yes this is high on my priority list. I plan to write a small tutorial on how to do this. In the meantime, I recommend checking out the [V-REP tutorial](http://www.coppeliarobotics.com/helpFiles/en/buildingAModelTutorial.htm) on how to model new robots, as this is a skill that is required in order to bring new robots in to PyRep also.

nanbaima, 2019-10-02:
I'm also really interested on that, I want to work with NAO model of vrep

stepjam, 2019-10-02:
Hi,
Yes I realize this is **well** overdue, especially since it shouldn't take very long. I'll put some time aside this weekend :)

stepjam, 2019-10-05:
Please find the tutorial [here](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md).

closed 2019-10-05 by stepjam

nanbaima, 2019-10-05:
> Please find the tutorial [here](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md).

Thanks stepjam, it will help a lot.
Just to make it clear, the model of NAO has a naming problem, what would be your advice? Change the name of joints and other thing on the scene? on the model? or adapt the classes to the ¨official¨ name? (as it already come implemented on the v-rep repo)

nanbaima, 2019-10-21:
> Thanks stepjam, it will help a lot.
> Just to make it clear, the model of NAO has a naming problem, what would be your advice? Change the name of joints and other thing on the scene? on the model? or [...]

Hy Stepjam,
Just to give a feedback, I followed your instructions and I renamed the NAO model joints and structure. It worked great!
However I would like to work with the NAO (hand) manipulator, how to implement new objects such as hands and legs on humanoids robots? Any tips on that? :see_no_evil: 
Great work you have been doing! :astonished:

obasekore, 2021-10-30:
> > > Please find the tutorial [here](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md).
> > 
> > 
> > Thanks stepjam, it will help a lot.
> > Just to make it clear, the model of NAO has a naming problem, what would be your advice? Change the name of joints and other thing on the scene? on the model? or adapt the classes to the ¨official¨ name? (as it already come implemented on the v-rep repo)
> 
> Hy Stepjam, Just to give a feedback, I followed your instructions and I renamed the NAO model joints and structure. It worked great! However I would like to work with the NAO (hand) manipulator, how to implement new objects such as hands and legs on humanoids robots? Any tips on that? 🙈 Great work you have been doing! 😲

Hey @nanbaima,
I also attempted to add NAO to PyRep for my experiment by following the tutorial while building inspiration from baxter robot.

Joy fully It was successful, **but unfortunately** the IK and path generation were returning errors. Either something unable to find path or target too far. And I obviously placed the targets/waypoints within the workspace, in fact close very close to the tip.

I'm really wonder if these errors has something to do with NAO's DoF, because most manipulator added on PyRep have full DoF (at least 6).

I am curious about your success on adding NAO, could you kindly assist ?

nanbaima, 2021-11-03:
> > > > Please find the tutorial [here](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md).
> > > 
> > > 
> > > Thanks stepjam, it will help a lot.
> > > Just to make it clear, the model of NAO has a naming problem, what would be your advice? Change the name of joints and other thing on the scene? on the model? or adapt the classes to the ¨official¨ name? (as it already come implemented on the v-rep repo)
> > 
> > 
> > Hy Stepjam, Just to give a feedback, I followed your instructions and I renamed the NAO model joints and structure. It worked great! However I would like to work with the NAO (hand) manipulator, how to implement new objects such as hands and legs on humanoids robots? Any tips on that? see_no_evil Great work you have been doing! astonished
> 
> Hey @nanbaima, I also attempted to add NAO to PyRep for my experiment by following the tutorial while building inspiration from baxter robot.
> 
> Joy fully It was successful, **but unfortunately** the IK and path generation were returning errors. Either something unable to find path or target too far. And I obviously placed the targets/waypoints within the workspace, in fact very close to the tip.
> 
> I really wonder if these errors has something to do with NAO's DoF, because most manipulator added on PyRep have full DoF (at least 6).
> 
> I am curious about your success on adding NAO, could you kindly assist ?

Hey @obasekore, how are you?

Well, I got it to work with this [pull](https://github.com/stepjam/PyRep/pull/196) request, however, I never implemented the IK path generator.
I later saw that the NAO joints had a few misplacements and some configurations never worked well because there was no consistency in the setup on the VRep model. I mean, there is the model right, buuuttt we need to improve it.
So my tip to you would be to double, no, actually, triple check the setup and all little boxes checking and testing everything. (really this took a lot of time and made me mistake a lot)

I hope my pull request might help you, I've never finished it because Stephan asked me to implement the walking example and I had no time to implement it, as I was working with the upper part, feel free to try and test it. I got my master's thesis with it. :)

commit 5f777602 (2019-10-05): [ci skip] Resolves #38. Tutorial on adding robots.