stepjam/PyRep #40: How to read the distance of a registered distance object?

kuroko1789, 2019-09-05:
Hi Stephen,

I try to read the distance of a distance object with `vrep.simReadDistance`, but the return value is 3.4028232762376986e+36 instead of  the value it should return which is 0.6.

stepjam, 2019-09-05:
Hi,
It looks like I haven't really wrapped the distance object functionality, I can add this in today or tomorrow (hence why you had to use `vrep.` instead of `pr.` etc.). Regardless, what you are showing should work, so I will look in to it. Can you send me your (.ttt) scene file please?

- Stephen

stepjam, 2019-09-05:
Hi,
The issue was that you needed to call [simHandleDistance](http://www.coppeliarobotics.com/helpFiles/en/regularApi/simHandleDistance.htm).
Regardless, I recommend you now use the `pyrep.misc.distance` module. Please pull and reinstall.

-Stephen

closed 2019-09-05 by stepjam

stepjam, 2019-09-05:
Worth noting that you can also get the distance between 2 objects using [check_distance](https://github.com/stepjam/PyRep/blob/master/pyrep/objects/object.py#L607).
e.g.
```
ob1 = Shape('ob1')
ob2 = Shape('ob2')
dist = ob1.check_distance(ob2)
```
-Stephen

lgzid, 2023-07-22:
Hi,
Can you provide some examples of using the pyrep.msc.distance module?
I don't understand how to use it.

commit cb3e0c36 (2019-09-05): Resolves #40. Added distance object.