stepjam/PyRep #41: Problem when importing a class outside the main program

bdelhaisse, 2019-09-05:
Hi,

I have a problem when loading/importing a class that uses the simulator in another file.
To illustrate the problem, here is my first simplified Python file `test1.py`:

```python
from pyrep import PyRep
from pyrep.objects.shape import Shape, PrimitiveShape

class TestShape(object):
    def __init__(self):
        self.pr = PyRep()
        self.pr.launch('scene.ttt', headless=False)
        self.pr.start()

    def create_box(self):
        shape = Shape.create(type=PrimitiveShape. color=[1, 0, 0, 0.5], size=[1], 
                             position=[0, 0, 2])

    def step(self):
        self.pr.step()

    def close(self):
        self.pr.stop()
        self.pr.shutdown()

if __name__ == '__main__':
    test = TestShape()
    test.create_box()
    for i in range(1000):
        test.step()
    test.close()
```
which, if run, works as expected. You can see a red sphere in the V-REP simulator.

However, when I import the class from another file like `test2.py`:
```python
from test1 import TestShape

test = TestShape()
test.create_box()
for i in range(1000):
    test.step()
test.close()
```
The sphere disappear in the simulator... Well, technically it is still there but at the position (x=NaN, y=NaN, z=NaN) according to V-Rep... 

How to fix this?

stepjam, 2019-09-05:
Hi,

That's very strange, there seems to be nothing wrong. In fact wrapping the sim code in a class like this is exactly how I expect most people to be using PyRep.

Is what you have shared the only thing that you are running? I.e.  are you running the `TestShape` in a separate thread/process?

bdelhaisse, 2019-09-05:
Hum... That is strange, I tested it on another computer and it works fine. I will check what is going wrong on the other computer and if I find what was the problem, I will report it here.

stepjam, 2019-09-05:
Hmm, OK. Is there anything particularly special about the machine where it doesn't work? i.e. are you using a different OS, Python env, etc?

stepjam, 2019-09-17:
Hi @bdelhaisse. Any updates on this?

stepjam, 2019-09-26:
Closing due to inactivity.

closed 2019-09-26 by stepjam