stepjam/PyRep #42: Problem when calling simReadVisionSensor

kuroko1789, 2019-09-06:
Hi,
I want to get the measurement from SICK S300 Fast, so I call  `vrep.simReadVisionSensor(handle)` and get the following error
```
for i in range(auxValuesCount[0]):
TypeError: '_cffi_backend.CData' object cannot be interpreted as an integer
```
Is this a bug or I am doing something wrong?
Btw, will it be slower if I just run the accosicated child script of SICK S300 Fast, and get the measurement data by calling `sim.getStringSignal` ?

Coppelia, 2019-09-09:
Hello,
one approach would be to acquire the data you want from that sensor by slightly modify the child script attached to the sensor, so that it broadcasts the data as a string signal, then read the data via sim.getStringSignal. Another approach would be to call a function in that child script that returns the data read by the child script of the sensor. That would also involve slightly modifying the attached child script

stepjam, 2019-10-17:
Thanks @Coppelia :)
Closing due to inactivity.

closed 2019-10-17 by stepjam