stepjam/PyRep #48: import error

0xfish, 2019-09-18:
I keep getting this error when I try run the examples:

[jason@archlinux PyRep]$ python examples/example_locobot_stack_cube.py
Traceback (most recent call last):
  File "examples/example_locobot_stack_cube.py", line 8, in <module>
    from pyrep import PyRep
  File "/home/jason/.local/lib/python3.7/site-packages/pyrep/__init__.py", line 1, in <module>
    from .pyrep import PyRep
  File "/home/jason/.local/lib/python3.7/site-packages/pyrep/pyrep.py", line 1, in <module>
    from pyrep.backend import vrep, utils
  File "/home/jason/.local/lib/python3.7/site-packages/pyrep/backend/vrep.py", line 2, in <module>
    from ._v_rep_cffi import ffi, lib
ImportError: /home/jason/.local/lib/python3.7/site-packages/pyrep/backend/_v_rep_cffi.cpython-37m-x86_64-linux-gnu.so: undefined symbol: simExtGetExitRequest

stepjam, 2019-09-18:
Hi,
I'm pretty sure that your issue will be solved if you navigate to any other directory and then run the example (for example, navigate to the examples folder and run the script). 

The reason why this happens is because it is using the local version of PyRep rather than the installed version.

Please confirm if this solves your issue.

stepjam, 2019-09-26:
Closing due to inactivity.

closed 2019-09-26 by stepjam