stepjam/PyRep #49: How to know if robot tip achieved target?

esperancaleonardo, 2019-09-18:
I'm doing some RL testes using Panda arm, and would like to know how to detect if the robot gripper tip achieved the target. I'm using this, and wanna know if this is right:

`done = (Env.actuator.get_tip().get_position()[i] == Env.target.get_position()[i] for i in range(3) )`

Is this right or do I have another/better way of doing that?

stepjam, 2019-09-18:
Hi,
Yes you are on the right track. But remember, that the tip position will never *exactly* be equal to the target. So you will need to have some distance threshold that you set. You can see an example of this in the examples folder. https://github.com/stepjam/PyRep/blob/master/examples/example_reinforcement_learning_env.py

Alternatively, I like to place proximity sensors in my scene and then use those to detect if the tip has entered the target. This is nice because you can visualise what your distance threshold looks like by changing the volume of your proximity sensor.

esperancaleonardo, 2019-09-20:
Thanks!! what range do you think would be good? like a 0.1 amount? more/less?

stepjam, 2019-09-26:
Sorry for delay. This depends on what task you are trying to do.
But for a reaching task, yes that seems sensible :)

closed 2019-09-26 by stepjam