stepjam/PyRep #51: cannot find -lv_rep error

mohsen0311, 2019-09-23:
i just define every thing and setup requirements but after Troubleshooting i still have this error on install the setup.py  here it is what consoled after the command :
`creating symlink: /mnt/c/Program Files/v-rep3/v-rep_PRO_EDU/libv_rep.so.1 -> /mnt/c/Program Files/v-rep3/v-rep_PRO_EDU/libv_rep.so
copying lua file: pyrep/backend -> /mnt/c/Program Files/v-rep3/v-rep_PRO_EDU
generating build/pyrep/backend/pyrep/backend/_v_rep_cffi.c
(already up-to-date)
not modified: 'build/pyrep/backend/pyrep/backend/_v_rep_cffi.c'
running install
running build
running build_py
package init file 'pyrep/robots/mobiles/__init__.py' not found (or not a regular file)
package init file 'pyrep/robots/mobiles/__init__.py' not found (or not a regular file)
running build_ext
building 'pyrep.backend._v_rep_cffi' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/mnt/c/users/mohsen/desktop/testlin/pyrep/cffi_build -I/usr/include/python3.6m -c build/pyrep/backend/pyrep/backend/_v_rep_cffi.c -o build/temp.linux-x86_64-3.6/build/pyrep/backend/pyrep/backend/_v_rep_cffi.o
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.6/build/pyrep/backend/pyrep/backend/_v_rep_cffi.o -L/mnt/c/Program Files/v-rep3/v-rep_PRO_EDU -lv_rep -o build/lib.linux-x86_64-3.6/pyrep/backend/_v_rep_cffi.cpython-36m-x86_64-linux-gnu.so
/usr/bin/ld: cannot find -lv_rep
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1`

mohsen0311, 2019-09-23:
`path = os.path.join(os.environ['VREP_ROOT'], 'libv_rep.so')
if not os.path.exists(path + '.1'):
    print('creating symlink: %s -> %s' % (path + '.1', path))
    os.symlink(path, path + '.1')` 
i got the error on spyder in these lines

sholtodouglas, 2019-09-29:
Second issue only occurs after you try compilation once, you can fix it with 'rm libv_rep.so.1'. I also got the first issue and don't know how to fix it.

mohsen0311, 2019-09-29:
yes i found it mu self but it dosent work on windows as i found out (im beginner to this) but it actually work on ubuntu just run it there if anu other library needed it will show in terminal to install it

closed 2019-09-29 by mohsen0311