stepjam/PyRep #53: camera and vision access

mohsen0311, 2019-09-29:
for accsesing the camera and vision senseors we should pyrep or import vrep and use the api of vrep for python?

esperancaleonardo, 2019-09-29:
Mohsen0311 you can directly use pyrep, see its docs on How to do It, there is also lots of examples from examples folder.

stepjam, 2019-09-30:
Hi,
You should use the [VisionSensor class (pyrep/objects/vision_sensor.py)](https://github.com/stepjam/PyRep/blob/master/pyrep/objects/vision_sensor.py).

You can do something like the following:

```python
from pyrep import PyRep
from pyrep.objects.vision_sensor import VisionSensor

pr = PyRep()
# Launch the application with a scene file in headless mode
pr.launch('scene.ttt', headless=True) 
pr.start()  # Start the simulation

vs = VisionSensor('my_vision_sensor')  # The name of the sensor in the vrep scene
rgb = vs.capture_rgb()
depth = vs.capture_depth()
# ... do some stuff ...
pr.shutdown()
```

closed 2019-09-30 by stepjam