stepjam/PyRep #55: Cannot use cv2.imshow()

remi2257, 2019-10-04:
Hi Stephen and V-REPers,

Just to give you the context, I'm using V-REP to simulate a robot which sort objects and I want to use OpenCV to do the computer vision, so that I can command the robot from Python directly after the image process.

My issue is that I cannot display what robot sees with a classic "cv2.imshow()", and it's made the development quite hard because I'm "blind". 
(Btw, cv2imwrite() is working, and I can confirm that simulation is running and capture_rgb() is working as expected)

The code is : 
```
from os.path import dirname, join, abspath
import cv2
import numpy as np
from pyrep import PyRep
from pyrep.objects.vision_sensor import VisionSensor

SCENE_FILE = join(dirname(abspath(__file__)), 'my_simu.ttt')

pr = PyRep()
# Launch the application with a scene file in headless mode
pr.launch(SCENE_FILE, headless=True)
pr.start()  # Start the simulation

vs = VisionSensor('Vision_sensor_hand')  # The name of the sensor in the vrep scene

# while True:
for i in range(100):
    pr.step()

    # -- Finding Instant Const -- #
    rgb = np.uint8(vs.capture_rgb() * 255) # Capture image and put values between 0 and 255

    #depth = vs.capture_depth()

    # Processing
    # cv2.imwrite('save/{:04d}.jpg'.format(i), rgb)
    cv2.imshow('image raw', cv2.cvtColor(rgb))

    k = cv2.waitKey(1) & 0xFF
    if k == 27:
        break

pr.stop()  # Stop the simulation
pr.shutdown()  # Close the application
```
The execution leads to 
```
Error: signal 11:

/home/remi/Documents/V-REP_PRO/libv_rep.so.1(_Z11_segHandleri+0x2b)[0x7fab73f7a4ab]
/lib/x86_64-linux-gnu/libc.so.6(+0x3ef20)[0x7faba11a5f20]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtCore-6df570dd.so.4.8.7(_ZN16QCoreApplication12libraryPathsEv+0x5b)[0x7fab9a3ea1cb]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtCore-6df570dd.so.4.8.7(_ZN14QFactoryLoader6updateEv+0x20)[0x7fab9a3cfd30]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtCore-6df570dd.so.4.8.7(_ZN14QFactoryLoaderC2EPKcRK7QStringN2Qt15CaseSensitivityE+0x15a)[0x7fab9a3d001a]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtGui-903938cd.so.4.8.7(+0x2646ab)[0x7fab9abeb6ab]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtGui-903938cd.so.4.8.7(_ZN19QApplicationPrivate18x11_apply_settingsEv+0x1cd5)[0x7fab9abf1c55]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtGui-903938cd.so.4.8.7(+0x26cd2d)[0x7fab9abf3d2d]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtGui-903938cd.so.4.8.7(+0x2701a5)[0x7fab9abf71a5]
/home/remi/.local/lib/python3.6/site-packages/cv2/.libs/libQtGui-903938cd.so.4.8.7(_ZN19QApplicationPrivate9constructEP9_XDisplaymm+0xc8)[0x7fab9ab8aa98]
QMutex: destroying locked mute
```

Have you ever met this issue and is there any solution which can overcome it ?

Cheers :smile:

stepjam, 2019-10-05:
Hi @remi2257.
I was indeed able to reproduce this. It looks like there is a conflict between opencv and some of the QT functionality; this only seems to effect the `imshow` function. In the meantime, perhaps display images with Pillow instead?

remi2257, 2019-10-05:
Hey @stepjam ,
I didn't try Pillow yet, but I did try with Matplotlib and the result was the same "Error : Signal 11"
I will test Pillow on Monday and I will keep in touch for any alternative !

Have a good WE

remi2257, 2019-10-07:
Just a little update,

I didn't find a convenient way to use Pillow for this type of use, so I'm currently using PyGame to display camera's streaming.

Cheers

mghayour, 2020-11-01:
i have same problem

lianghongzhuo, 2020-11-01:
@mghayour you can try to use  Headless version of opencv package: 
`pip install opencv-python-headless`