stepjam/PyRep #56: No module named 'tests' when trying to test new robots

nanbaima, 2019-10-07:
Hey stepjam,
Once again, great work. 
Now I´m implementing the NAO robot, and right after following your instructions, when running the `test_arms_and_configuration_paths.py`, I´m getting the error > no module named ´tests´. I´ve installed PyRep as you suggested:
`pip3 install -r requirements.txt
python3 setup.py install --user`
Do you have any suggestions?

NataliaSanchezTamayo, 2019-10-16:
I am having the same issue. @nanbaima did you find a way to solve it?

nanbaima, 2019-10-16:
> I am having the same issue. @nanbaima did you find a way to solve it?

Well, I didn't fixed it persay, but I did got it to work by basically copying the whole class of TestCore inside the file `core.py` (that is at the same file base)  and pasting inside the `test_arms_and_configuration_paths.py` code. It worked, and I got to get the unittests outputs.

Just to give you a hint: when I tried to run/excute the test I was initially trying to run in terminal by being inside the `PyRep/tests` files. By that I kept getting the `ModuleNotFoundError: No module named 'pyrep.backend._v_rep_cffi'  `, so after a lot of time spading on that, I found on [Troubleshooting](https://github.com/stepjam/PyRep#install), that I had to run it outside the project root.

Hope it helps you! :smile:

stepjam, 2019-10-16:
Hi,
Have you tried running:
`python -m unittest test_arms_and_configuration_paths.py<http://test_arms_and_configuration_paths.py>`

-Stephen

Sent from my phone (please excuse any typos!)

NataliaSanchezTamayo, 2019-10-21:
Thanks, that worked!