stepjam/PyRep #59: How to get script parameters and signals ?

remi2257, 2019-10-15:
Hi Stephen,

I have created a pick & place scene, and I'm using signal and script parameters to share informations between scripts.
To illustrate my problem : I'm choosing, via an UI on V-Rep, which type of can will be generated by my generator, and then I load the model & send information about can's shape through a signal.
With PyRep, I'm kind of blocked because I cannot get this Signal or even Load the can's model and look for shape, because shape info is a ScriptParameter.

Did I miss something ? Is there a solution ? Or could you add these functions to PyRep :smile: ?

Cheers

stepjam, 2019-10-18:
Yes this can be added to PyRep quite easily. Can you list the exact functions you are interested in? See here: http://www.coppeliarobotics.com/helpFiles/en/apiFunctionListAlphabetical.htm

remi2257, 2019-10-18:
Great !

On signals, it's mostly Set/Get Operations:

sim.setIntegerSignal
sim.getIntegerSignal
sim.clearIntegerSignal
sim.setDoubleSignal
sim.getDoubleSignal
sim.clearDoubleSignal
sim.setStringSignal
sim.getStringSignal
sim.clearStringSignaj
sim.getSignalName 

On ScriptParameter:

sim.setScriptSimulationParameter
sim.getScriptSimulationParameter

Once you implemented it, I will check out your commit, so that, I can just merge request it next time :) 

Merci & Bonne journée à toi ;)

Rémi

stepjam, 2019-10-19:
In terms of timeline, I'm off to Japan tomorrow, so not sure if I will have time to add in these additions before then. Will be back in 2 weeks to add it in if not though :).

-Stephen

stepjam, 2019-10-19:
Oops, didn't mean to close.

stepjam, 2019-12-15:
Sorry for the delay in getting this in. 
Signaling is now in. I couldn't seem to find the script simulation parameters in the new CoppeliaSim, so `sim.setScriptSimulationParameter` and `sim.getScriptSimulationParameter` have not been fully added; although I have now wrapped the c++ functions, so when that feature becomes available again in CoppeliaSim we can get these added in quickly.

closed 2019-12-15 by stepjam

remi2257, 2019-12-15:
Niiiiiice

Hoping that this new version of V-REP will lead to better performances for easier AI 😄 

Thx again and cheers!

xman236, 2021-05-17:
Is there any update regarding `setScriptSimulationParameter, ``getScriptSimulationParameter`?

commit 8464e5ac (2019-12-15): Resolves #59. Added signaling.