stepjam/PyRep #60: Question and missing "add the _ik element" on tutorial

nanbaima, 2019-10-16:
Hi Stepjam,
I was following your instructions on [Adding Robots Tutorial](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md), and I think you might had forgotten to explain to click add Ik elements on `Kinematics group`, I'm not sure but, the unittest  only worked after adding it to the new added ik group.

About the tutorial I have a question about `Dummys`, over there you say to:
> In the scene hierarchy, the 'target' dummy should be a child of the robot root node, and the tip dummy should be a child of the last node in the robot arm. Position both the dummies at the tip of the arm, and orient the tip dummy such that the z-axis points out from the arm.
Why is that?
I did it, but when I started my simulation on my experiment, the 'target' dummy kept being lost and being in the midle of nowhere, trying to follow the tip, but like it had an delay. So I putted it as an child of the last node in the robot arm.
Is it wrong? Should I be concerned about anything?

Once again, great work!
Thanks!

stepjam, 2019-10-17:
Hi,
Yes you are right. I'll add that detail to the tutorial. Thanks for spotting that!
Regarding the second point: I guess it doesn't matter where you put the target as in the backend, pyrep moves the target when you use `get_path`, `get_linear_path`, etc.

closed 2019-10-17 by stepjam

nanbaima, 2019-10-21:
> Hi,
> Yes you are right. I'll add that detail to the tutorial. Thanks for spotting that!
> Regarding the second point: I guess it doesn't matter where you put the target as in the backend, pyrep moves the target when you use `get_path`, `get_linear_path`, etc.

Hey,
Oohh OK, got it!
So I will keep them together.
Thanks.

commit dc255b52 (2019-10-17): [ci skip] Resolves #60. Additional steps in robot tutorial.