stepjam/PyRep #61: Slow simulation on headless machine with xvfb

quantumiracle, 2019-10-18:
Hi Stephen, 

I found that the speed of learning program with PyRep environments to be significantly lower on headless machine using 'xvfg-run' as you suggested, compared with running on PC with screen. (5-10 times slower)

Do you have any idea to speed it up? Thanks.

Best,
Zihan

stepjam, 2019-10-19:
Hi Zihan,

Yes Xvfb does not use the gpu to render, which is probably why you are getting the slowdown (I should update the readme).
Try using TigerVNC instead for the virtual framebuffer.

Best,
Stephen

quantumiracle, 2019-10-19:
Thanks, Stephen!

quantumiracle, 2019-10-25:
Hi Stephen, 

I've tried with using TigerVNC instead of Xvfb, but it seems the process is still slow. I compare the launching logs for headless mode on my PC and on server, there is only one difference about the 'Image' plugin saying: 
```
Plugin 'Image': loading...
Error with plugin 'Image': load failed (could not load). The plugin probably couldn't load dependency libraries. For additional infos, modify the script 'libLoadErrorCheck.sh', run it and inspect the output.
```
which does not affect the program to successfully launch but may be the reason of the slow process. 
I'm not sure if this is the reason, or I may not use tigervnc correctly. What I did was using a local client to display the virtual desktop of the server with tigervnc, and then run the script normally with `python **.py`. Do you have any idea on solving that?

Best regards,
Zihan

stepjam, 2019-10-31:
Hi Zihan,

I'm currently away, so can't look in to this at the moment. Perhaps take a look at how CARLA (driving simulator) does this. Looks like they use turbovnc. See here:
https://github.com/carla-simulator/carla/blob/master/Docs/carla_headless.md
https://github.com/carla-simulator/carla/issues/225

Let me know how you get on.

stepjam, 2019-12-14:
Hi Zihan, were you able to get anywhere with this?

quantumiracle, 2019-12-24:
Hi Stephen, 

Sorry I haven't noticed your reply before. 

I didn't keep tracking this problem since last reply, but I managed to install the 'Image' package. 

Recently I tested PyRep on the server again, and it seems there is no more problem regarding the speed of simulation (the server is still a little bit slower, but not that significant anymore, considering the different configurations of server and my PC, I would say there should be no problem). And I tested with both TigerVNC and 'xvfb-run', the speeds are similar.

I guess it may be caused by some package dependence problem (maybe I changed some packages version in later usage) or the server itself, but I don't know exactly what happened. But thanks for your patience and replies!

Best,
Zihan

closed 2019-12-24 by quantumiracle

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless