stepjam/PyRep #62: Install error: fatal error: pyconfig.h: No such file or directory

IanYangChina, 2019-10-21:
Hi, Stephen, I followed the instruction and I got the following error
I'm not sure what's going on.

**PyRep$ python3 setup.py install --user**

**Output:**
copying lua file: pyrep/backend -> /home/xintong/Documents/Apps/V-REP/
generating build/pyrep/backend/pyrep/backend/_v_rep_cffi.c
(already up-to-date)
not modified: 'build/pyrep/backend/pyrep/backend/_v_rep_cffi.c'
running install
running build
running build_py
package init file 'pyrep/robots/mobiles/__init__.py' not found (or not a regular file)
package init file 'pyrep/robots/mobiles/__init__.py' not found (or not a regular file)
running build_ext
building 'pyrep.backend._v_rep_cffi' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/xintong/PyRep/cffi_build -I/usr/include/python3.6m -c build/pyrep/backend/pyrep/backend/_v_rep_cffi.c -o build/temp.linux-x86_64-3.6/build/pyrep/backend/pyrep/backend/_v_rep_cffi.o
build/pyrep/backend/pyrep/backend/_v_rep_cffi.c:22:24: **fatal error: pyconfig.h: No such file or directory**
compilation terminated.
**error: command 'x86_64-linux-gnu-gcc' failed with exit status 1**

IanYangChina, 2019-10-23:
Problem solved, I tried sudo apt-get install python3-dev but it didn't work.
And I realized that I am actually using python 3.6 rather than the preinstalled 3.5.
So I tried python3.6-dev, then it works.

closed 2019-10-23 by IanYangChina

yliu634, 2021-12-08:
works for me!