stepjam/PyRep #64: Multiple PyRep() instances

guilhermecano, 2019-10-24:
Hi. I do research in RL and, in order to run an algorithm, I must instantiate two environments/simulations at the same time. However, when I instantiate and launch two PyRep objects, the first scene starts correctly but the second one doesn't seem to start the simulation. The errors are also different when headless is set to False and when it is set to true.

Headless True (for the second vrep only):
```
WARNING: QApplication was not created in the main() thread.
(python:10027): GLib-CRITICAL **: g_main_context_push_thread_default: assertion 'acquired_context' failed
```
Then, the second simulation doesn't start.

Headless False:
```
00003008 $$$ERROR in uiThread_tryToLockForUiEventRead (_handleMouseAndKeyboardAndResizeEvents): not called from the UI thread.
QObject::setParent: Cannot set parent, new parent is in a different thread
```

It seems the threads are conflicting somehow. Do you know of any solution for this problem? Is it currently possible to instantiate and launch multiple PyRep() objects?

stepjam, 2019-10-31:
Hi,
It sounds like you are running threads rather than processes. Please see: https://github.com/stepjam/PyRep#running-multiple-pyrep-instances.

Please reopen if the problem persists. 

Best,
Stephen

closed 2019-10-31 by stepjam

fuying-wang, 2020-08-27:
Hi, @stepjam, I am wondering if `PyRep` could support running multiple threads?  Actually, I want to do some reinforcement learning work, and I am trying to integrate this environment to OpenAI gym. However, I'm stuck here.