stepjam/PyRep #66: Error from running the example code: pyrep.errors.ConfigurationPathError

BoyuanChen, 2019-10-26:
Hi Stephen,

Thanks for the great work!

I am getting the following error when I try to run the example code for locobot. Could you please take a look at it? Thank you!

Using the 'MeshCalc' plugin.
Initializing the Bullet physics engine in plugin 'DynamicsBullet_2_83'...
Engine version: 2.83.7
Plugin version: 10
Initialization successful.
Driving to cube ...
Move arm above cube ...
Traceback (most recent call last):
  File "examples/example_locobot_stack_cube.py", line 63, in <module>
    move_arm(grasp_point_raised, grasp_point.get_quaternion())
  File "examples/example_locobot_stack_cube.py", line 40, in move_arm
    ignore_collisions=ignore_collisions)
  File "/home/cml/env3.6/lib/python3.6/site-packages/pyrep/robots/arms/arm.py", line 276, in get_path
    trials_per_goal, algorithm)
  File "/home/cml/env3.6/lib/python3.6/site-packages/pyrep/robots/arms/arm.py", line 234, in get_nonlinear_path
    raise ConfigurationPathError('Could not create path.')
pyrep.errors.ConfigurationPathError: Could not create path.
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex

Thanks!!
Bo

BoyuanChen, 2019-10-26:
It looks like sometimes the nonlinear path cannot be computed even though the program starts with the scene. I only get like twice success out of ten trials. Is this caused by the physical noise? Are there other ways to make this more robust?

stepjam, 2019-12-18:
Yes this is due to noise. I probably need to spend some time tweaking the example (e.g. by slightly moving the waypoint/cube). The workspace of the locobot arm is small, so there's not much margin for error.

BoyuanChen, 2019-12-21:
Ah, got it. Thanks Stephen!

closed 2020-07-03 by BoyuanChen