stepjam/PyRep #67: Test Robotiq85 gripper: 0.3324768383074185 != 0.8 within 0.05 delta

IanYangChina, 2019-10-26:
Hi Stephen,

I added a Robotiq 85 gripper into the test_scene_robots.ttt in the assets folder.
Followed every step in the 'adding robot' tutorial, python files etc.
And it passed the get_gripper, duplicate and copy tests.

But it couldn't pass the close and open test, throwing an AssertionError
......
**gripper.get_open_amount()[0], 0.8, delta=0.05)**
**File "/usr/lib/python3.6/unittest/case.py", line 878, in assertAlmostEqual**
**raise self.failureException(msg)**
**AssertionError: 0.3324768383074185 != 0.8 within 0.05 delta**

The gripper velocities I tried were 0.04 and 0.2, which were those you set for testing other grippers.

And when testing it also gave me a warning: QApplication was not created in the main() thread.

I don't know what is going wrong.
I've compared the properties of this gripper with other pre-built ones, they look all the same.

Thanks!

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been away from the office for ~2 months.

I guess the first thing to check is if gripper is opening when you look at it in the scene. I.e. ceate a small python file that starts V-REP (not in headless) and then fetch the gripper, and call `my_gripper.actuate(0.8)`. You should see that it opens 80% of the way. If it doesn't then please share your findings.

IanYangChina, 2019-12-18:
Thanks for replying Stephen, but I'm no longer working on the Vrep environment now and probably won't have time to try your suggestion out...
So you may close this issue or leave it to someone else.
Cheers mate

closed 2019-12-18 by stepjam