stepjam/PyRep #70: get_position() returning NAN

pbustos, 2019-11-07:
Hello Stephen,

I am using the UR10 arm in a gym custom environment and with the stable-baselines algorithms and after a few hundred epochs the agent.get_position() call returns NAN values in all coordinates, and keeps that way after. 
I can't find an explanation for this behaviour. Have you seen it before?

Also, I would like to ask you about activating the shadows when rendering with your version of V-REP.

Thanks,

Pablo

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been out of office for ~2 months.

Thanks for reporting. I have not seen this behaviour. Does the `get_position()` call get the position of the end-effector? I'm not sure how to debug this. Could you perhaps try and get a minimal working example that you could send me? A loop that randomly moves the arm and then make sire that it crashes after a certain amount of time?

If it happens as infrequently as every few hundred steps, then perhaps in the short term you can wrap each call with a check to see if it's Nan, and if it is, then restart the environment such that it doesn't crash the program?

Regarding your second question, yes you can use the OpenGL3 renderer. You can set this via:
```
from pyrep.objects.vision_sensor import VisionSensor
from pyrep.const import RenderMode

vs = VisionSensor('my_vs')
vs.set_render_mode(RenderMode.OPENGL3)
```

pbustos, 2019-12-18:
Thanks for answer.
I'll try to reproduce the error in a simple setup. I asked because get_position retrieves the angles of the joints in the manipulator using PyRep method for that. Once I get the first vector of NANs, all other readings are NAN and the arm does not seem to be in a singular configuration or even a strange position. 

And thanks also for the answer to the GL3 renderer.

Pablo

stepjam, 2020-02-10:
Hi @pbustos 
Have you managed to reproduce?

pbustos, 2020-02-24:
Hi,

no, that project is stopped for now. We are modelling the Kinova Gen3 to continue our research with both the simulator and the real arm. We'll submit a pull request we are done.

Thanks

closed 2020-07-04 by stale[bot]

lgzid, 2023-07-29:
@pbustos 
Hi
I have also encountered the same problem recently. Every time I train for a certain number of epochs, get_ position() returns Nan, how can I solve it