stepjam/PyRep #73: Not able to use notebook or ipython for interactive sessions

Tete-Xiao, 2019-11-17:
Hey folks,

This is a really nice work! I'm trying to use the `embed` function in IPython for creating interactive sessions so that it's easier to probe and play with the environment. However, after the environment starts simulation, I'm not able to call embed function, as the following error will appear:

`Traceback (most recent call last):
  File "tmp.py", line 52, in <module>
    embed()
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/IPython/terminal/embed.py", line 387, in embed
    frame.f_code.co_filename, frame.f_lineno), **kwargs)
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/traitlets/config/configurable.py", line 412, in instance
    inst = cls(*args, **kwargs)
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/IPython/terminal/embed.py", line 159, in __init__
    super(InteractiveShellEmbed,self).__init__(**kw)
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/IPython/terminal/interactiveshell.py", line 460, in __init__
    self.init_prompt_toolkit_cli()
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/IPython/terminal/interactiveshell.py", line 292, in init_prompt_toolkit_cli
    **self._extra_prompt_options())
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/prompt_toolkit/shortcuts/prompt.py", line 344, in __init__
    output = output or get_default_output()
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/prompt_toolkit/output/defaults.py", line 67, in get_default_output
    return create_output()
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/prompt_toolkit/output/defaults.py", line 45, in create_output
    stdout, term=get_term_environment_variable())
  File "/home/txiao/anaconda3/lib/python3.7/site-packages/prompt_toolkit/output/vt100.py", line 420, in from_pty
    isatty = stdout.isatty()
ValueError: I/O operation on closed file`

More specifically, I tried `example_turtlebot_navigation.py` in PyRep. As long as I insert embed after simulation starts, e.g., in the action loop, the program throws out the error. Even at the end of program I had this error.

Thanks!

stepjam, 2019-12-14:
Hi Tete. Apologies for the delay in getting back to you on this issue, I have been away from the office for ~2 months.

I'm afraid I don's use IPython and not sure what you mean by 'insert embed'.
Could you perhaps give more information?

Thanks!

wkentaro, 2020-02-06:
I think this issue is related to https://github.com/stepjam/PyRep/pull/113#issuecomment-582597456, since `get_path()` also uses `with suppress_std_out_and_err()` inside.
The simplest solution, for now, is using below embedding of pure python instead of other third-parties (e.g., pdb, ipdb, ipython).

```python
import pyrep

def main():
    pr = pyrep.PyRep()
    pr.launch()

    ...

    local = globals().copy()
    local.update(locals())
    import code; code.interact(local=local)  # go into interactive shell

    # XXX: don't use followings
    # import pdb; pdb.set_trace()
    # import ipdb; ipdb.set_trace()
    # import IPython; IPython.embed()

...
```

wkentaro, 2020-02-06:
Also, I found https://github.com/stepjam/PyRep/pull/118 solves the issue on IPython.

commit 2981898c (2020-02-07): Check if the code is running in IPython to avoid I/O Error (#118)

Resolves #73.
closed 2020-02-07 by stepjam