stepjam/PyRep #74: ImportError: libv_rep.so.1 when launching from terminal

bsangiovanni, 2019-11-19:
I can not get PyRep to work. The installation procedure goes smoothly but when I launch any example script from **terminal** (NO PYCHARM) I get this.
`ImportError: libv_rep.so.1: cannot open shared object file: No such file or directory`

How can I solve this problem?

InspireX96, 2019-11-20:
Seems that they removed `libv_rep.so.1` in the latest release of V-REP 3.6.2.
In my case, I created a soft link under V-REP folder:
```
ln -s libv_rep.so libv_rep.so.1
```
and it worked for me.

jianye0428, 2019-11-22:
Hello, 

I'm trying to install PyRep in my Laptop. My OS is Ubuntu 16.04. When i finished and run the code below:
[code]from pyrep import PyRep[/code]

A error arises:

libv_rep.so.1: cannot open shared object file: No such file or directory

I have googled about this problem and try to figure it out with the code under the folder VREP:
[code]ln -s libv_rep.so libv_rep.so.1 [/code]

but it didn't work, and the file libv_rep.so and libv_rep.so.1 already existed in folder VREP.

How can I solve the problem ?

Jian

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been away from the office for ~2 months.

This is related to **V-REP** changing to **CoppeliaSim**. I hadn't had time to transfer PyRep over, until today. The problem you are having should hopefully go away after updating, so I'm closing for now. Feel free to re-open if the problem is not solved after following the new README. Please make sure you first correctly remove V-REP and uninstall PyRep, before following install instructions.

closed 2019-12-14 by stepjam