stepjam/PyRep #75: Is it possible that reset many time could cause some error accumulated?

kuroko1789, 2019-11-20:
Hi Stephen,

I have used pioneer p3dx for training mobile robot to navigate using reinforcement learning. Everything looked normal at first, but the reward decreased dramatically after running the simulation for a while. So I try to save the model and the replay buffer and restart the program once in a while, the reward doesn't drop dramatically this time and the end result is quite good. Is it possible the dynamics simulation will accumulate errors? If yes, is there a way to fix it?

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been away from the office.

Thanks for raising. Could you send me a simple working example (.ttt  and .py file) where this behaviour emerges? Maybe something like a loop that gets the position, moves forward for a few steps, resets, and then prints the error? We can then see if this error accumulation is indeed happening.

Thanks

stepjam, 2020-02-10:
@kuroko1789  Any updates?

kuroko1789, 2020-02-11:
Hi, 

I haven't got the time to look in to it again. I will reply you these days.

P-Schumacher, 2020-03-16:
Hey, I also seem to have this problem. In my case it is because I have to use a somewhat unstable gripper model. The model tends to break after colliding multiple times with another object. 
What is the fastest way to reset the model in such a case? If I just reset the positions, the gripper stays broken. I could pr.stop() and pr.start() the whole simulation and then reacquire the handles, but that is slow.
Any ideas?

Ramomrito, 2020-03-21:
Hi @P-Schumacher,

You can save the configuration tree of your gripper model using: `initial_gripper_configuration = gripper.get_configuration_tree()` at the start of your training loop. Then you can reset the configuration tree with `pr.set_configuration_tree(initial_gripper_configuration)` every time the model breaks. The configuration tree basically holds all the relative transformations between a parent and its children.

P-Schumacher, 2020-03-21:
Thanks it worked perfectly and is much faster!

closed 2020-07-04 by stale[bot]