stepjam/PyRep #76: A problem about PyRep  version `Qt_5.12' not found

jianye0428, 2019-11-22:
hello,
 
I want to use PyRep to control the object in Vrep and run the code below:
[code]from pyrep import PyRep[/code]

and a problem arises:

/home/jianye/anaconda3/lib/libQt5Core.so.5: version `Qt_5.12' not found (required by /usr/lib/libv_rep.so.1)

any advice?

Thanks in advance!

Jian Ye

kaelgabriel, 2019-11-27:
"/home/jianye/anaconda3/lib/libQt5Core.so.5:" your anaconda env is linking to its own qt binaries.

First create an env just for pyrep!

1) conda list | grep 'qt'

2) Check what packages do you have with qt

3) conda remove --force <INSERT_EACH_PACKAGE> (--force only removes the package and not its dependencies)

4) try to run it again

DarthThomas, 2019-12-04:
I got a similar import error:

`ImportError: /home/tom/proj/robotics/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/libcoppeliaSim.so.1: symbol __cxa_throw_bad_array_new_length, version Qt_5 not defined in file libQt5Gui.so.5 with link time reference`

I'm using:

qt                        5.12.1               hbfa2a6e_0    conda-forge

kaelgabriel, 2019-12-04:
http://www.coppeliarobotics.com/previousVersions

Try V-REP 3.6.2 Versions

DarthThomas, 2019-12-10:
sorry for the delay,  I'll try it tomorrow

DarthThomas, 2019-12-11:
tried  V-REP 3.6.2 Version, got same error:/home/tom/robotics/V-REP_PRO_EDU_V3_6_2_Ubuntu16_04/libv_rep.so.1: symbol __cxa_throw_bad_array_new_length, version Qt_5 not defined in file libQt5Gui.so.5 with link time reference

kaelgabriel, 2019-12-11:
Hi Darth, did you deleted your conda QT packages?

DarthThomas, 2019-12-12:
Hi Kael, thanks for the quick reply! Did you mean delete all the qt packages in my conda env for Pyrep?
I have this for `conda list | grep qt` in Pyrep env(12h ago):
```
pyqt                      5.12.3           py37hcca6a23_1    conda-forge
pyqtwebengine             5.12.1                   pypi_0    pypi
qt                        5.12.5               hd8c4c69_1    conda-forge
```
And I just removed them expecting to tell you still doesn't work. But Oops, it works now! Thanks!

But I'm just wondering why this work: Is it because somehow pyrep should be using qt in conda base or somewhere else, and the error appears when qt installed in the current environment?

kaelgabriel, 2019-12-12:
I'm glad it worked! I think is because conda envs are at the beginning of your PATH variable, so they look for the binaries there. But it is just a guess.

stepjam, 2019-12-14:
Given that the issue seems to be resolved, I am closing. 

Of note, there is now a new version of PyRep that I heavily recommend that you switch to. It uses the new CoppeliaSim rather than V-REP.

closed 2019-12-14 by stepjam

Gloriabhsfer, 2019-12-15:
@DarthThomas 
Hi Thomas: 
I have got the same issue but when I run the 

> conda list | grep 'qt'

it seems my qt version is 5.9.7 and I cannot find anaconda package for qt5.12 online
Can you tell me how to update your package?

Thanks
Gloria

DarthThomas, 2019-12-17:
> @DarthThomas
> Hi Thomas:
> I have got the same issue but when I run the
> 
> > conda list | grep 'qt'
> 
> it seems my qt version is 5.9.7 and I cannot find anaconda package for qt5.12 online
> Can you tell me how to update your package?
> 
> Thanks
> Gloria

Hi Gloria!

You don't need to update them, you just need to remove what you've got in your current conda env(assuming you are not using base env).

BTW I used `conda install -c conda-forge qt=5.12.1 `if I remember correctly.

stepjam, 2019-12-18:
Hi @DarthThomas :
PyRep and Conda issues seem to be common. Would you perhaps consider making a pull request for troubleshooting with conda in the README?

Best,
Stephen

DarthThomas, 2019-12-20:
> Hi @DarthThomas :
> PyRep and Conda issues seem to be common. Would you perhaps consider making a pull request for troubleshooting with conda in the README?
> 
> Best,
> Stephen

Hi Stephen!

I'll make a new conda env and try Pyrep set up with the newest version(CoppeliaSim) first.

I'd love to make a pull request about troubleshooting, but I have to make sure I understand it first.

I probably will do that next week and get back to you!

Cheers!
Thomas

DarthThomas, 2019-12-26:
> Hi @DarthThomas :
> PyRep and Conda issues seem to be common. Would you perhaps consider making a pull request for troubleshooting with conda in the README?
> 
> Best,
> Stephen

Hi @stepjam 

I made a new conda env and tried to set up the whole thing. Encountered the same qt problem. But this time just removing qt in the env didn't solve the problem... Currently, my solution is to clone the env worked with V-REP3.6.2 earlier, and rerun the `python setup.py install --user `.

But during the process, I noticed one potential reason for the problem, which is the python version. using python 3.7.5 seems to solve the problem. I'll test and verify later this week or next week.

I'll start writing tips for conda once I can set up the whole thing in a new env successfully.

Thomas

thanard, 2020-11-17:
I followed  the suggestion by @kaelgabriel; however, I ran into a different issue after removing all conda qt packages.

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/thanard/anaconda3/lib/python3.7/site-packages/pyrep/__init__.py", line 6, in <module>
    from .pyrep import PyRep
  File "/home/thanard/anaconda3/lib/python3.7/site-packages/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/home/thanard/anaconda3/lib/python3.7/site-packages/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ImportError: /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5: version `Qt_5' not found (required by /home/thanard/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu16_04/libcoppeliaSim.so.1)`