stepjam/PyRep #79: VREP_ROOT defined, but is not a valid path.

jianye0428, 2019-11-25:
hello,
I install the PyRep according to the tutorials. and I get this problem:

File "setup.py", line 2, in <module>
    import cffi_build.cffi_build as cffi_build
  File "/home/jianye/PyRep/cffi_build/cffi_build.py", line 12, in <module>
    raise RuntimeError('VREP_ROOT defined, but is not a valid path.')
RuntimeError: VREP_ROOT defined, but is not a valid path.

and my path setting is:
export VREP_ROOT=/home/jianye/vrep
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$VREP_ROOT
export QT_QPA_PLATFORM_PLUGIN_PATH=$VREP_ROOT

What is the problem ?
jianye

remi2257, 2019-11-28:
1) Did you add these paths to your bashrc and then source it ?
2) Are VREP's files directly in "/home/jianye/vrep" on not in a subfolder like "/home/jianye/vrep/VREP_EDU_UBUNTU18"

Cheers

stepjam, 2019-12-14:
Thanks @remi2257, these are good suggestions.

Also of note, **V-REP** has changed to **CoppeliaSim**. So @jianye0428 , I recommend removing V-REP and following the new instructions for installation. I am closing, please re-open if you have any problems.

closed 2019-12-14 by stepjam