stepjam/PyRep #82: How to get robot end-effector with PyRep

Gloriabhsfer, 2019-12-01:
Hi all:

I want to do a reinforcement learning with robot impedance control with PyRep.
In real environment, I got force-sensor to measure my end-effector force, and I want to get the force in v-rep. So I add the force-sensor at end-effector, and I can get the information back through the force sensor. But the type of force-sensor is Tuple[List[float],List[float]]. Ideally, I can get a list of 2 tuple with 3 force and 3 torque. But I get the wrench information like this:

> ValueError: could not convert string to float: '0,[0.0, 0.0, 0.0],[0.0, 0.0, 0.0]'
And when I calculate the joint target velocity through these force, I need to convert the tuple into real number. 
Now I feel very confuse. I can get the force and torque seprately ,but I cannot use these number as my control input because the type issue. And I searched on remote api of VREP, they didn't tell me what's 

> FORCE_SENSOR = vrep.sim_object_forcesensor_type
Can anyone tell me how to modify the force_sensor.py in objects and make the output of forcesensor as a string of real number and read them in my controller?
Thanks again for the brillant package.

Best,
Gloria

stepjam, 2019-12-14:
Hi,

I'm not sure I fully understand the question. Could you send me exactly what you are trying to run, what outputs you are getting, and what you are expecting. 
I assume you are doing something like:
```
# ... Setup sim ...
# ...
from pyrep.objects.force_sensor import ForceSensor
my_sensor = ForceSensor('my_sensor')
forces_xyz, torques_xyz = my_sensor.read()
```

Gloriabhsfer, 2019-12-15:
@stepjam 
Hi Stephen:

Thanks for your reply. That's exactly what I want. Thanks for your help.

Best,
Gloria

closed 2019-12-15 by Gloriabhsfer