stepjam/PyRep #84: version Qt_5 not defined in file libQt5Gui.so.5 with link time reference

hetolin, 2019-12-05:
Hi,when I run example`python example_baxter_pick_and_pass.py`
**I got the error:**
`ImportError: /home/hetolin/vrep/libv_rep.so.1: symbol __cxa_throw_bad_array_new_length, version Qt_5 not defined in file libQt5Gui.so.5 with link time reference`

I have install qt and qt5 in my anaconda env:
Name                      Version                   
pyqt                          5.12.3          
pyqt5-sip                  4.19.18                
pyqtwebengine         5.12.1                   
qt                              5.12.5               
qt5                            5.3.1                        
qtconsole                  4.6.0                  

and I googled this problem, but the method provided did not work.
How can I solve such problem?

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been away from the office for ~2 months.

This is related to **V-REP** changing to **CoppeliaSim**. I haven't had time to transfer PyRep over, until today. The problem you are having should hopefully then go away, so I'm closing for now. Feel free to re-open if the problem is not solved after following the new README. Please make sure you first correctly remove V-REP and uninstall PyRep, before following install instructions.

closed 2019-12-14 by stepjam

marcoewerton, 2020-03-26:
Hello. I get the same error as hetolin. I am using CoppeliaSim_Edu_V4_0_0_Ubuntu16_04.

marcoewerton, 2020-03-26:
It works now. I've followed the suggestions that have been given here:
https://github.com/stepjam/PyRep/issues/76