stepjam/PyRep #85: I got a similar import error:

hetolin, 2019-12-06:
Hi, Have you solve this problem?I got a same import error as you:
I tried many method,but they did not work

`ImportError: /home/hetolin/coppeliasim/libcoppeliaSim.so.1: symbol __cxa_throw_bad_array_new_length, version Qt_5 not defined in file libQt5Gui.so.5 with link time reference`

I'm using:

qt                        5.12.5      
qt5 　　　　　　5.3.1
_Originally posted by @DarthThomas in https://github.com/stepjam/PyRep/issues/76#issuecomment-561509484_

hetolin, 2019-12-06:
I uninstall qt5, and it works now,qt and qt5 cannot be installed simultaneously.

DarthThomas, 2019-12-10:
Thanks for the information, will try tomorrow!

DarthThomas, 2019-12-11:
I uninstalled qt5,  didn't work still get: 

`/home/tom/robotics/V-REP_PRO_EDU_V3_6_2_Ubuntu16_04/libv_rep.so.1: symbol __cxa_throw_bad_array_new_length, version Qt_5 not defined in file libQt5Gui.so.5 with link time reference`

`conda list | grep qt`
gives:
```
pyqt                      5.12.3           py37hcca6a23_1    conda-forge
pyqtwebengine             5.12.1                   pypi_0    pypi
qt                        5.12.5               hd8c4c69_1    conda-forge
```
any difference from you?

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been away from the office for ~2 months.

This is related to **V-REP** changing to **CoppeliaSim**. I haven't had time to transfer PyRep over. The good news is that I'm doing that today, and main branch should be updated by end of the day.
The problem you are having should then go away, so I'm closing for now. Feel free to re-open if the problem is not solved after following the new README.

closed 2019-12-14 by stepjam

stepjam, 2019-12-14:
Update is now live.