stepjam/PyRep #87: Crash starting up PyRep+CoppeliaSim

arkose, 2019-12-12:
Hello! I found two issues, one which I was able to get past, the second still causing CoppeliaSim to crash on loading.

I am using `CoppeliaSim_Edu_V4_0_0_Ubuntu18_04`. I have installed PyRep as instructed on the github page. I get an error saying that no such file `libcoppeliaSim.so.1` exists; it in fact does not exist in `CoppeliaSim_Edu_V4_0_0_Ubuntu18_04`, but `libcoppeliaSim.so` does; I assume this is an oversight on Coppelia's part.

```
$ python3 example_baxter_pick_and_pass.py 
Traceback (most recent call last):
  File "example_baxter_pick_and_pass.py", line 9, in <module>
    from pyrep import PyRep
  File "(...)/.local/lib/python3.6/site-packages/pyrep/__init__.py", line 1, in <module>
    from .pyrep import PyRep
  File "(...)/.local/lib/python3.6/site-packages/pyrep/pyrep.py", line 1, in <module>
    from pyrep.backend import vrep, utils
  File "(...)/.local/lib/python3.6/site-packages/pyrep/backend/vrep.py", line 2, in <module>
    from ._v_rep_cffi import ffi, lib
ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory
```

Copying `libcoppeliaSim.so` into `libcoppeliaSim.so.1` allows CoppeliaSim to start loading, with the splash screen displaying; however, it crashes on loading:

```
$ python3 example_baxter_pick_and_pass.py 
lib: 1
lic: 1

CoppeliaSim Edu V4.0.0. (rev. 1)
Using the default Lua library.
Loaded the video compression library.
Add-on script 'simAddOnScript-addOnScriptDemo.lua' was loaded.
Add-on script 'simAddOnScript-b0RemoteApiServer.lua' was loaded.
Add-on script 'simAddOnScript_PyRep.lua' was loaded.
Add-on script 'vrepAddOnScript_PyRep.lua' was loaded.
If CoppeliaSim crashes now, try to install libgl1-mesa-dev on your system:
>sudo apt install libgl1-mesa-dev
OpenGL: NVIDIA Corporation, Renderer: Quadro P1000/PCIe/SSE2, Version: 4.6.0 NVIDIA 430.26
...did not crash.
Simulator launched.
Plugin 'Geometric': loading...
Plugin 'Geometric': load succeeded.
Plugin 'Assimp': loading...
Initialization failed (registerScriptStuff): simRegisterScriptVariable: error
Assimp: failed to register script stuff
Error with plugin 'Assimp': load failed (failed initialization).
Plugin 'BlueZero': loading...
Initialization failed (registerScriptStuff): simRegisterScriptVariable: error
BlueZero: failed to register script stuff
Error with plugin 'BlueZero': load failed (failed initialization).
Plugin 'BubbleRob': loading...
Plugin 'BubbleRob': load succeeded.
Plugin 'Bwf': loading...

Error: signal 11:

(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x2b)[0x7f5da0568eab]
/lib/x86_64-linux-gnu/libc.so.6(+0x3ef20)[0x7f5da2c67f20]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so(_ZN19CSimAndUiThreadSyncD1Ev+0x23b)[0x7f5d4984229b]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so(_Z25simSetModuleInfo_internalPKciS0_i+0xa3)[0x7f5d49663ce3]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libsimExtBwf.so(simStart+0x1fa)[0x7f5d425b090a]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN7CPlugin4loadEv+0xa3)[0x7f5da076ef53]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN16CPluginContainer9addPluginEPKcS1_+0x110)[0x7f5da0775c60]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN9CUiThread27__executeCommandViaUiThreadEP16SUIThreadCommandS1_+0x211e)[0x7f5da06fe4be]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN9CUiThread25executeCommandViaUiThreadEP16SUIThreadCommandS1_+0x85)[0x7f5da06ff185]
(...)/Desktop/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_Z22simLoadModule_internalPKcS0_+0x3ba)[0x7f5da056cc2a]
QMutex: destroying locked mutex
```

Do you have any idea what might be the problem?

Thank you very much! I appreciate the time spent on this.

arkose, 2019-12-12:
Solution:

Instead of copying `libcoppeliaSim.so` into a new file `libcoppeliaSim.so.1`, I created a soft link called `libcoppeliaSim.so.1` linking to `libcoppeliaSim.so`, and now things load beautifully.

If I may, I request that instructions to create a soft link be added to the main github page under installation instructions, so others don't run into this problem.

closed 2019-12-12 by arkose

stepjam, 2019-12-14:
Hi. Apologies for the delay in getting back to you on this issue, I have been away from the office for ~2 months.

This is related to **V-REP** changing to **CoppeliaSim**. I haven't had time to transfer PyRep over, until today. The problem you are having should hopefully then go away, so I'm closing for now. Feel free to re-open if the problem is not solved after following the new README. Please make sure you first correctly remove V-REP and uninstall PyRep, before following install instructions.