stepjam/PyRep #88: Feature Request: Octree Support

arkose, 2019-12-14:
Hello! Thank you for all the hard work on PyRep.

I would like to humbly request that Octree support be added to PyRep.

These are the foreign functions that would need to be wrapped in some manner:

```
simInt simCreateOctree(...);
simInt simInsertVoxelsIntoOctree(...);
simInt simRemoveVoxelsFromOctree(...);
const float* simGetOctreeVoxels(...);
simInt simInsertObjectIntoOctree(...);
simInt simSubtractObjectFromOctree(...);
simInt simCheckOctreePointOccupancy(...);
```

Thank you!

stepjam, 2019-12-15:
Hi arkose. Yes this would be nice.
I can certainly do this, but I cant give an estimate as to when this would be done. Your best bet would be to have a go at wrap them yourselves and then send a PR that I can check over. You can see lots of examples of this in the sim.py file, along with test examples in the tests/ folder.

All the best,
Stephen

stepjam, 2020-02-10:
Feature in #91

closed 2020-02-10 by stepjam