stepjam/PyRep #92: UR3 and RG2 Path Planning

bibibibibit, 2019-12-25:
My arm is UR3 and my gripper is RG2.
I set collections (both arm and gripper) and IK_group (only UR3_tip and base is UR3). Also, I linked UR3_target with UR3_tip. And I put UR3_tip as the son of gripper. UR3_target as the son of arm. 
However, when I called the arm.get_path(), it told me cannot create the path correctly. What's the relationship between the two dummies above and new created waypoint dummies?

Thank you!

lromanini, 2020-02-07:
Same problem for me. Even giving a simple position (really near to the tip of the arm) i get the error message "Could not find path".
Is there something other to add/modify in the scene or in the code?
thanks

Zheyu-Zhuang, 2020-04-28:
Hey,
I've had this issue. I found a way to get around it. 
I would suggest first checking whether pyrep can solve the IK given the target pose. If it can solve ik, then try **toggle the check collision flag in the get_path function to False**. Mine works this way.

Cheers

closed 2020-07-04 by stale[bot]