stepjam/PyRep #94: Solved: ImportError: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.12' not found

matwilso, 2020-01-08:
I was getting an error with Qt when trying to run PyRep

`ImportError: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version 'Qt_5.12' not found (required by /home/matwilso/Downloads/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1)
`

I think this is because I had a different version of Qt5 installed elsewhere in my system.  I fixed this issue by reordering the LD_LIBRARY_PATH setting so that it would look for the Coppelia version first.  

`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:COPPELIASIM_ROOT`
to
`export LD_LIBRARY_PATH=COPPELIASIM_ROOT:$LD_LIBRARY_PATH`

I am not sure if this is the best fix, since it might lead to errors when running the other Qt programs.  But anyway, it solves the issue.

DarthThomas, 2020-01-13:
I tried this but didn't solve my problem. I still need to remove qt packages installed in my current conda env...

pandyarohan, 2020-03-31:
I was using anaconda and doing this worked for me: https://github.com/stepjam/PyRep/issues/76

closed 2020-07-04 by stale[bot]

bibbygoodwin, 2020-09-16:
@DarthThomas , I wonder if you just copy-pasted @matwilso 's line? If your `.bashrc` is set up as per the repo's guidance, eg:
`export COPPELIASIM_ROOT=EDIT/ME/PATH/TO/COPPELIASIM/INSTALL/DIR
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT
export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT
`
Then there's a missing `$` in the new line, and it should read:
`export LD_LIBRARY_PATH=$COPPELIASIM_ROOT:$LD_LIBRARY_PATH`

For me, changing this order was enough to fix the problem, and I couldn't do #76 as I'm on a Cluster where I cannot edit base conda packages.

DarthThomas, 2020-09-22:
> @DarthThomas , I wonder if you just copy-pasted @matwilso 's line? If your `.bashrc` is set up as per the repo's guidance, eg:
> `export COPPELIASIM_ROOT=EDIT/ME/PATH/TO/COPPELIASIM/INSTALL/DIR`
> `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT`
> `export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT`
> 
> ...then there's a missing `$` in the new line as proposed, and it should read:
> `export LD_LIBRARY_PATH=$COPPELIASIM_ROOT:$LD_LIBRARY_PATH`
> 
> For me, changing this order was enough to fix the problem, and I couldn't do #76 as I'm on a Cluster where I cannot edit base conda packages.

Hey! Thanks for the reply! Unfortunately, I changed the order, and it didn't work for me... I've been working on another project since the outbreak of COVID-19, will be back and keep trying when I finish. 
 :)  Again, thanks for reply!