stepjam/PyRep #95: error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

JMAgundezG, 2020-01-09:
While I was installing PyRep, this problem appeared to me and I can't compile.

```
XMNA13 :: ~/PyRep » python3 setup.py install --user
creating symlink: /home/jmagundezg/PyRep/libcoppeliaSim.so.1 -> /home/jmagundezg/PyRep/libcoppeliaSim.so
copying lua file: pyrep/backend -> /home/jmagundezg/PyRep
generating build/pyrep/backend/pyrep/backend/_sim_cffi.c
(already up-to-date)
not modified: 'build/pyrep/backend/pyrep/backend/_sim_cffi.c'
running install
running build
running build_py
package init file 'pyrep/robots/mobiles/__init__.py' not found (or not a regular file)
package init file 'pyrep/robots/mobiles/__init__.py' not found (or not a regular file)
running build_ext
building 'pyrep.backend._sim_cffi' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/jmagundezg/PyRep/cffi_build -I/usr/include/python3.6m -c build/pyrep/backend/pyrep/backend/_sim_cffi.c -o build/temp.linux-x86_64-3.6/build/pyrep/backend/pyrep/backend/_sim_cffi.o
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.6/build/pyrep/backend/pyrep/backend/_sim_cffi.o -L/home/jmagundezg/PyRep -lcoppeliaSim -o build/lib.linux-x86_64-3.6/pyrep/backend/_sim_cffi.cpython-36m-x86_64-linux-gnu.so
/usr/bin/ld: no se puede encontrar -lcoppeliaSim
collect2: error: ld returned 1 exit status
```

JMAgundezG, 2020-01-10:
The problem was the COPPELIASIM_ROOT source.

closed 2020-01-10 by JMAgundezG

AlaAlimi, 2023-01-27:
Hi, I am facing the same issue, could you please share you solved this problem