stepjam/PyRep #98: ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory

XuankeShi, 2020-01-13:
we i run the example
it occurs that RuntimeError: COPPELIASIM_ROOT defined, but is not a valid path.

stepjam, 2020-01-13:
Hi,
When running the _examples_, are you in the examples folder, rather than the _pyrep_ folder?

XuankeShi, 2020-01-13:
> RuntimeError: COPPELIASIM_ROOT defined, but is not a valid path.

> Hi,
> When running the _examples_, are you in the examples folder, rather than the _pyrep_ folder?

yes，of course

XuankeShi, 2020-01-13:
[image]

XuankeShi, 2020-01-13:
[image]

stepjam, 2020-01-13:
Ahh I see you are using conda. There are ongoing issues with this; see #93 #94.

XuankeShi, 2020-01-13:
i have try the python3.7 and change the envirment variable.  but it dosen't work.

stepjam, 2020-01-13:
Hmm, I'm not sure. Have you followed the updated readme in #93?
Let's sanity check something; can you run/install outside conda for now and check if that works?

XuankeShi, 2020-01-13:
i try to install outside conda,but it also failed.  such a bad news,i am so sad;
[image]

raghada, 2020-01-15:
Same here, I'm not using conda either

stepjam, 2020-01-15:
Are you able to run CoppeliaSim as normal (without Python)?

raghada, 2020-01-15:
Yes

stepjam, 2020-01-15:
I see the problem in the case of the last screenshot posted by @XuankeShi. You were running under python(2) rather than python3. Perhaps this is your issue too @raghada?

raghada, 2020-01-16:
No, I'm using python3

raghada, 2020-01-16:
[image]

XuankeShi, 2020-01-16:
the same question

XuankeShi, 2020-01-16:
> I see the problem in the case of the last screenshot posted by @XuankeShi. You were running under python(2) rather than python3. Perhaps this is your issue too @raghada?

i use python3 ,the screenshot i post may be a mistake.

raghada, 2020-01-16:
I'm using V4.0.0 Rev 4

raghada, 2020-01-17:
I tried reinstalling ubuntu as well as CoppeliaSim, no thing works.

stepjam, 2020-01-17:
I have some good news. I was able to reproduce by pulling the latest CoppeliaSim version. I had V4.0.0 Rev 2 locally. I can't fix this right now, but should be able to tomorrow (as long as the fix is a simple as I imagine).

For now, I recommend that you use revision 2 rather than revision 4.

raghada, 2020-01-17:
Very much appreciated!

raghada, 2020-01-17:
Coulod you please advice on where to find revision 2? I was not able to find it in their website

closed 2020-01-19 by stepjam

xysong1201, 2022-06-15:
Hi guys remember to `source ~/.bashrc` after you change the path, this solves the problem on my side..

Murphy103, 2022-09-29:
launch pycharm in terminal probably can solve this problem,because Pycharm could load LD_library_path from bashrc when you launch pycharm in terminal.

whc688, 2023-09-18:
> launch pycharm in terminal probably can solve this problem,because Pycharm could load LD_library_path from bashrc when you launch pycharm in terminal.

this works for me

commit 40e598e4 (2020-01-19): Update for CoppeliaSim 4.0.0.Rev 4. Resolves #98.