stepjam/PyRep #100: Lua runtime error: ~/vrep_3.6.2/socket.lua:13: module 'socket.core' not found when launching simulation using PyRep from Python script

bsangiovanni, 2020-01-14:
Hi,

I have been encountering this issue when launching the simulation of a scene with PyRep. One object of said scene implements non-threaded script of a socket communication, for receiving signals from an external hardware. 
Said script has something like this

```
function sysCall_init()          
        HOST = <<IP_ADDRESS>>
        PORT = 2002 

        local socket=require("socket") 
        client=socket.tcp() 
 
        result=client:connect(HOST, PORT) 
       
        local dataOut = 'V-rep Connected' 
        writeSocketData(client,dataOut) 
 
        --local dataIn = readSocketData(client) 
        --print(dataIn) 
        --i = 0 
        local dataIn = readSocketData(client) 
        print(dataIn) 
end 
```

If I launch V-Rep normally (i.e., command line), it has no problem running such simulation (no error messages whatsoever). However, if I launch and start said simulation using PyRep from a Python script, I get the following error with a traceback up to 
`local socket=require("socket") `

```
Lua runtime error: ~/vrep_3.6.2/socket.lua:13: module 'socket.core' not found:
no field package.preload['socket.core']
	no file './socket/core.lua'
	no file '/usr/local/share/lua/5.1/socket/core.lua'
	no file '/usr/local/share/lua/5.1/socket/core/init.lua'
	no file '/usr/local/lib/lua/5.1/socket/core.lua'
	no file '/usr/local/lib/lua/5.1/socket/core/init.lua'
	no file '~/vrep_3.6.2/socket/core.lua'
	no file '~/vrep_3.6.2/lua/socket/core.lua'
	no file '~/vrep_3.6.2/bwf/socket/core.lua'
	no file '~/new_user/my_project/scenes/socket/core.lua'
	no file './socket/core.so'
	no file './libsocket/core51.so'
	no file '/usr/local/lib/lua/5.1/socket/core.so'
	no file '/usr/local/lib/lua/5.1/libsocket/core51.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'
	no file './socket.so'
	no file './libsocket51.so'
	no file '/usr/local/lib/lua/5.1/socket.so'
	no file '/usr/local/lib/lua/5.1/libsocket51.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'
stack traceback:
    [C]: in function 'require'
    ~/vrep_3.6.2/socket.lua:13: in main chunk
    [C]: in function 'require'
```

Any clue why this happens/how to solve it?

stepjam, 2020-01-16:
Hi,
Before we dive in, can I just check what version of CoppeliaSim you are using?
Thanks

bsangiovanni, 2020-01-18:
Apologies for the late reply: I am using the second to last v-rep version (3.6.2).

stepjam, 2020-01-19:
Please update to using the new CoppeliaSim release (V4.0.0). That may not fix the issue, but at least you will be up to date and then we can proceed from there :smile: .

If that does not fix the issue, then please post a minimal working example that reproduces the problem.

stepjam, 2020-02-01:
@bsangiovanni Was the issue resolved after updating?

closed 2020-07-04 by stale[bot]