stepjam/PyRep #101: Suppressing printout flood from CoppeliaSim

mightestDuck, 2020-01-16:
Hi

Whenever launching a scene or starting a simulation through PyRep the terminal gets flooded with messages from CoppeliaSim. I tried re-directing the stdout before calling a function (launch() or start()), as suggested [here](https://stackoverflow.com/questions/8391411/suppress-calls-to-print-python) but that didn't remove them. 

I start and stop simulation regularly which makes the relevant printouts get lost in this flood of useless information. Is there any way to suppress these messages?

PS. Thanks for the library, it makes my work a whole lot easier!

**Starting a simulation:**
```
Initializing the Bullet physics engine in plugin 'DynamicsBullet_2_83'...
Engine version: 2.83.7
Plugin version: 11
Initialization successful.
```

**Launching:**

```
lib: 1
lic: 1

CoppeliaSim Edu V4.0.0. (rev. 4)
Using the default Lua library.
Add-on script 'simAddOnScript-addOnScriptDemo.lua' was loaded.
Add-on script 'simAddOnScript-b0RemoteApiServer.lua' was loaded.
Add-on script 'simAddOnScript_PyRep.lua' was loaded.
Simulator launched.
Plugin 'Geometric': loading...
Plugin 'Geometric': load succeeded.
Plugin 'Assimp': loading...
Plugin 'Assimp': warning: replaced variable 'simAssimp'
Plugin 'Assimp': load succeeded.
Plugin 'BlueZero': loading...
Error with plugin 'BlueZero': load failed (could not load). The plugin probably couldn't load dependency libraries. For additional infos, modify the script 'libLoadErrorCheck.sh', run it and inspect the output.
Plugin 'BubbleRob': loading...
Plugin 'BubbleRob': load succeeded.
Plugin 'Bwf': loading...
Plugin 'Bwf': load succeeded.
Plugin 'CodeEditor': loading...
CodeEditor: cannot load in headless mode
Error with plugin 'CodeEditor': load failed (failed initialization).
Plugin 'Collada': loading...
CoppeliaSim runs in headless mode. Cannot start 'Collada' plugin.
Error with plugin 'Collada': load failed (failed initialization).
Plugin 'ConvexDecompose': loading...
Plugin 'ConvexDecompose': load succeeded.
Plugin 'CustomUI': loading...
CustomUI: doesn't work in headless mode
Error with plugin 'CustomUI': load failed (failed initialization).
Plugin 'DynamicsBullet-2-78': loading...
Plugin 'DynamicsBullet-2-78': load succeeded.
Plugin 'DynamicsBullet-2-83': loading...
Plugin 'DynamicsBullet-2-83': load succeeded.
Plugin 'DynamicsNewton': loading...
Plugin 'DynamicsNewton': load succeeded.
Plugin 'DynamicsOde': loading...
Plugin 'DynamicsOde': load succeeded.
Plugin 'DynamicsVortex': loading...
Plugin 'DynamicsVortex': load succeeded.
Plugin 'ExternalRenderer': loading...
Plugin 'ExternalRenderer': load succeeded.
Plugin 'ICP': loading...
Plugin 'ICP': warning: replaced variable 'simICP'
Plugin 'ICP': load succeeded.
Plugin 'Image': loading...
Error with plugin 'Image': load failed (could not load). The plugin probably couldn't load dependency libraries. For additional infos, modify the script 'libLoadErrorCheck.sh', run it and inspect the output.
Plugin 'K3': loading...
Plugin 'K3': load succeeded.
Plugin 'Lic': loading...
Plugin 'Lic': load succeeded.
Plugin 'LuaCommander': loading...
LuaCommander: cannot find the statusbar widget
Error with plugin 'LuaCommander': load failed (failed initialization).
Plugin 'LuaRemoteApiClient': loading...
Plugin 'LuaRemoteApiClient': load succeeded.
Plugin 'Mtb': loading...
Plugin 'Mtb': load succeeded.
Plugin 'OMPL': loading...
Plugin 'OMPL': warning: replaced variable 'simOMPL'
Plugin 'OMPL': load succeeded.
Plugin 'OpenGL3Renderer': loading...
Plugin 'OpenGL3Renderer': load succeeded.
Plugin 'OpenMesh': loading...
Plugin 'OpenMesh': load succeeded.
Plugin 'Qhull': loading...
Plugin 'Qhull': load succeeded.
Plugin 'RRS1': loading...
Plugin 'RRS1': load succeeded.
Plugin 'ReflexxesTypeII': loading...
Plugin 'ReflexxesTypeII': load succeeded.
Plugin 'RemoteApi': loading...
Starting a remote API server on port 19997
Plugin 'RemoteApi': load succeeded.
Plugin 'SDF': loading...
SDF: doesn't work in headless mode
Error with plugin 'SDF': load failed (failed initialization).
Plugin 'SurfaceReconstruction': loading...
Plugin 'SurfaceReconstruction': warning: replaced variable 'simSurfRec'
Plugin 'SurfaceReconstruction': load succeeded.
Plugin 'Urdf': loading...
CoppeliaSim runs in headless mode. Cannot start 'Urdf' plugin.
Error with plugin 'Urdf': load failed (failed initialization).
Plugin 'Vision': loading...
Plugin 'Vision': load succeeded.
Using the 'Geometric' plugin.
```

mightestDuck, 2020-01-16:
Solved, the approach I had tired before was suppressing the output produced within Python only. [This approach](https://stackoverflow.com/questions/4178614/suppressing-output-of-module-calling-outside-library) removes messages from outside too.

closed 2020-01-16 by mightestDuck

nanbaima, 2020-04-12:
> Solved, the approach I had tired before was suppressing the output produced within Python only. [This approach](https://stackoverflow.com/questions/4178614/suppressing-output-of-module-calling-outside-library) removes messages from outside too.

Hey @mightestDuck, I know this is not a big deal, but I confess that this output is annoying...
What exactly did you do? Becouse I still want to print a few things of mine and I want to only exclude every time this appears (which is every epoch, by the Bullet lib):
`Initializing the Bullet physics engine in plug-in 'DynamicsBullet_2_78'...
Engine version: 2.78
Plugin version: 11
Initialization successful.`

mightestDuck, 2020-04-12:
Hey @nanbaima, a re-usable solution involves a custom decorator:
```
import os
import sys

def suppress_printout(func) -> object:
    def wrapper(*args, **kwargs):
        with open('/dev/null', 'w') as devnull:
            oldstdout_fno = os.dup(sys.stdout.fileno())
            os.dup2(devnull.fileno(), 1)
            returned = func(*args, **kwargs)
            os.dup2(oldstdout_fno, 1)
            return returned
    return wrapper
```
When defining a function that you want to suppress:
```
@suppress_printout
def a_function():
    # coppelia_stuff    
    ...
```
All printouts generated by a_function() are dumped now.

nanbaima, 2020-04-13:
Thanks @mightestDuck, It works fine!
Great Job. Thanks :)
However it is more a work around then a real solution. Perhaps @Coppelia, could help us here. Is it possible to give the path to were it is printed (which lib)?

nanbaima, 2020-05-28:
Just to give an heads up,[ this solution](https://github.com/stepjam/PyRep/pull/173#issue-418987850) solves the problem!
Nicely done @djl11