stepjam/PyRep #103: Scene launched by PyRep is empty

timtody, 2020-01-28:
Hello,

I'mn using the latest version of CoppeliaSim Edu and PyRep but with the supplied demo script 

`from pyrep import PyRep

pr = PyRep()
# Launch the application with a scene file in headless mode
pr.launch('scene.ttt', headless=True) 
pr.start()  # Start the simulation

# Do some stuff
for _ in range(1000):
pr.step()

pr.stop()  # Stop the simulation
pr.shutdown()  # Close the application`

the resulting scene is always blank for multiple different ttt files I tried. When I open those files with CoppeliaSim directly, the scenes look just as excpected.

scene file: https://gofile.io/?c=IlMky8

stepjam, 2020-02-01:
Hi timtody,
Make sure you give the absolute path of the file as CoppeliaSim needs this. Please let me know if this resolves the issue.

stepjam, 2020-02-01:
No need to supply abs path now, PyRep does this now automatically.

commit 048cfd9b (2020-02-01): Absolute path passed to CoppeliaSim. Resolves #103.
closed 2020-02-01 by stepjam